openrocket/openrocket #2492: [Feature Request] Sort motor configurations alphabetically

OZ1SEJ, 2024-06-04:
**Is your feature request related to a problem? Please describe.**
When dealing with a lot of different motor configurations, it is unclear how the list is sorted, and clicking the column header ("Configuration") doesn't do anything;

**Describe the solution you'd like**
I would like to be able to sort the list alphabetically, either by default or by clicking the column header.

**Describe alternatives you've considered**
I believe the most viable alternative is to just not implement this feature.

**Additional context**
Screenshot attached here:

[image]

JoePfeiffer, 2024-06-04:
This would really be a nice feature

neilweinstock, 2025-02-03:
Yes, this has bugged me for a long time.  We should be able to use the same sortable table as the sim results.