openrocket/openrocket #2493: I can't save project

Errcode95, 2024-06-04:
---------- Bug report ----------

1. Made a model
2. Clicked save

Exception stack trace: https://pastebin.com/j1bxHuF4

[image]

JoePfeiffer, 2024-06-04:
I'm having trouble duplicating the error... Something I'm noticing in your screenshot is you appear to have a pod set with no children. That's not causing the error when I create a design with the same thing, but it shouldn't be there.

SiboVG, 2024-06-04:
Can you try to delete some components and verify whether you can save your file after the deletion?

Errcode95, 2024-06-04:
Я удалил разгонные блоки и всё заработало, но теперь я не могу их вернуть

SiboVG, 2024-06-04:
> Я удалил разгонные блоки и всё заработало, но теперь я не могу их вернуть
> 
> ![image](https://private-user-images.githubusercontent.com/108075602/336550432-a4dc2fbb-da0c-446b-acb2-54c655287f13.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1MjI2MDcsIm5iZiI6MTcxNzUyMjMwNywicGF0aCI6Ii8xMDgwNzU2MDIvMzM2NTUwNDMyLWE0ZGMyZmJiLWRhMGMtNDQ2Yi1hY2IyLTU0YzY1NTI4N2YxMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwNFQxNzMxNDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZTZlYzFjZjQ1NjZlN2I1MjAwMzc5ODU3MDY2NDg5YTZjNTQzZTlkMTFlY2ZkYTEwNTc2YmUxN2JmYjY1OTQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.AEe4QiodvRxA-bB5PkHjJZCuIp9GL1o-afwQ36OJsBA) (3D схема так не думает)

Translation: "I removed the booster stages and everything worked, but now I can't get them back"
"(3D model doesn't think so)"

@Errcode95 could you respond in English from now on? Makes it difficult for us to help you otherwise.

JoePfeiffer, 2024-06-05:
If you're able to save the .ork file now, please do so, change its name from rocket.ork (or whatever it is) to rocket.zip, and post it.  Looking at that screenshot it appears the internal data structures have gotten completely confused; I'm not sure how (and I'm not sure we'll be able to figure out how), but the .ork file could help.

JoePfeiffer, 2024-07-18:
It's been over a month with no further response.