openrocket/openrocket #2520: "Paint Shop" for easy coloring of rockets

neilweinstock, 2024-07-25:
I'm pretty sure this will never be implemented but I'm going to write it down anyway for posterity; maybe it will inspire someone.  No mockups provided because I think it's incredibly easy to visualize from the description, but I can provide if someone wants.

Even with all of the *wonderful* improvements to the Appearance UI in recent versions of OR, managing rocket appearance is still tedious and unpleasant.  Here is a way to make at least part of it incredibly easy (decals will always be difficult).  I'll call it "Paint Shop".

Paint Shop is a small UI (could be a floating palette or built into the main OR window) with the following controls:
1) Color selector
2) Shine selector

These work as in the existing appearance panel to select a color and shine.

Then there are two buttons that are equivalent to the analogous tool selectors in paint programs:

3) Eyedropper:  When this button is activated, the cursor changes to an eyedropper.  Clicking on any rocket component will cause its appearance settings to be loaded into the Color and Shine selectors.

4) Spray Can: When this button is activated, the cursor changes to a spray can.  Clicking on any rocket component will set its appearance to the existing Color and Shine selectors.

Hitting the "Escape" key would cancel either of the above tools.

That's the very simple essence.  A few small wrinkles:
a) "Clicking on any rocket component" applies to anywhere in the UI you can select a component.  That would be the component tree, or any of the views in the Rocket Figure display.

b) Components that do not have an appearance (e.g., Pods) would not be clickable.

c) Components that can optionally have multiple appearances (i.e. fins and tubes) would require a bit of special handling.  I would imagine that clicking such a component would display a little contextual menu that lets you select which appearance you're referencing.  All the nitty-gritty details are TBD.

d) Not sure how to handle the eyedropper with components that are set to "default" appearance.  It's not particularly important but it would need to be handled in some sort of rational way.

That's it.

SiboVG, 2024-07-25:
It would probably fit best in the Tools menu. When using eye dropper and spray can tool, it would probably be good that the currently hovered component has an outline highlight, maybe even small textbox under the mouse with the component name, so you know which component you're about to click.

neilweinstock, 2024-07-25:
> It would probably fit best in the Tools menu.

That's the most likely option, probably, although I think it's interesting to think of it being directly in the main UI, collapsible so that it's only there when you need it.  But the tools menu would certainly work.

>When using eye dropper and spray can tool, it would probably be good that the currently hovered component has an outline highlight, maybe even small textbox under the mouse with the component name, so you know which component you're about to click.

Yes!  I meant to add that but forgot in all my frothy enthusiasm.

SiboVG, 2024-07-25:
For this to work on the 3D view for macOS, I'm afraid #754 will need to fixed first though...

neilweinstock, 2024-07-25:
> For this to work on the 3D view for macOS, I'm afraid #754 will need to fixed first though...

Yes, I am aware.  But even without that it'd be useful.  I could put up the side view and then keep the Photo Studio open, and get most of the same effect.