openrocket/openrocket #2521: Sim Plot tooltips do not show name of "sustainer" stage

neilweinstock, 2024-07-26:
I do not know if this was a conscious choice or if it's a bug.  Either way IMHO it should be changed.

When rolling over one of the data lines in a sim plot, a tooltip appears showing details.  For booster stages the name of the stage is shown at the top of the tooltip.  For the sustainer stage (including single-stage rockets) the name of the stage is *not* given.

I believe the stage name should be shown for *all* stages in multi-stage rockets.  It's debatable whether it should be shown for single-stage rockets, but I'd be inclined to include it.

JoePfeiffer, 2024-07-28:
I would argue it should only appear on multistage rockets, since there's no need to disambiguate a single stage. But yes, it should be added to the sustainer on a multistage plot.

neilweinstock, 2024-07-28:
I'm fine with that.

JoePfeiffer, 2024-07-28:
Related -- when showing events as icons, the icon tooltip doesn't show the stage name. This should be consistent with the data tooltip.

NCoppola25, 2024-08-02:
Hi, I would like to work on this issue.

SiboVG, 2024-08-02:
> Hi, I would like to work on this issue.

Great, thanks!

NCoppola25, 2024-08-02:
Any advice on where to get started? This is my first time contributing, but I've been trying to understand the codebase and how everything is connected (its quite a bit to go through). My intuition is that the sustainer stage is having its name left out of the tooltip dialogue because its the first stage, and my understanding is that for single-stage rockets the name is not shown in the tooltip. I could be totally wrong though!

JoePfeiffer, 2024-08-03:
I don't know your experience with github, so I may be telling you something here that you already know, but start by forking the repository, cloning it to your workstation, and making sure you are able to build the program. Most of our developers (I think all but me, actually) use IntelliJ as their IDE; I work from the command line.

Take a look at the developer's guide at https://openrocket.readthedocs.io/en/latest/ (this documentation also goes into much more detail on how to get started). There is a codebase diagram at https://openrocket.readthedocs.io/en/latest/dev_guide/codebase_walkthrough.html which will hopefully be helpful.

The specific place in the code I'd point you would be swing/src/main/java/info/openrocket/swing/gui/plot/SimulationPlot.java. That's where the actual plot is constructed, so that's where the tooltips are generated.

And please, keep asking questions if you need help!

NCoppola25, 2024-08-03:
Thanks so much! That is really helpful. I'll reach out if I have any other questions.

NCoppola25, 2024-08-05:
I just wanted to confirm - the desired behavior here is to include the name "Sustainer" for the portion of the main curve after all boosters have disconnected? Rather than have it show "Sustainer" for the entire main curve when the booster(s) are still connected? For example, on the altitude curve below.
[image]

neilweinstock, 2024-08-05:
> I just wanted to confirm - the desired behavior here is to include the name "Sustainer" for the portion of the main curve after all boosters have disconnected? Rather than have it show "Sustainer" [...]

1) Just to be clear, it should use the name assigned to the sustainer, not "Sustainer" (although that is the default name it can be changed)

2) You raise a good point about the part of the curve before separation; I hadn't noticed that it was unlabeled.  I think if I had my choice I would want it to be labeled with the booster name, because that is the "active" stage before separation.  Only after separation is the Sustainer active.  The fact that it is currently unlabeled makes me concerned that it will be difficult to do that, but I'm not sure.

What do you think, @JoePfeiffer @jimmiedave @SiboVG?

JoePfeiffer, 2024-08-06:
I actually hadn't thought about it.  After seeing these comments, I wrote (and then deleted) justifications for why there should be no label for that part of the flight (no disambiguation needed), why it should be "Sustainer" (easiest given how sim branches are created), and why it should be "Booster" (expected by people watching launch coverage on youtube).

Now I'm thinking it should be "Sustainer+Booster1+Booster2"

NCoppola25, 2024-08-06:
Currently I have a solution for showing the "Sustainer" name for the entire main curve (a small edit to 1 line of code). I also have a solution for showing no label up until the final booster separates, and then showing the "Sustainer" name during the portion where it is active (a bit more involved, but the only changes are to the SimulationPlot class). 

If it is preferred to show the specific "Booster" that is active during each portion, or to show all stages that are present for each portion of the main flight, then just let me know.

Thanks!

JoePfeiffer, 2024-08-07:
@neilweinstock @jimmiedave @SiboVG 
Comments?

SiboVG, 2024-08-07:
Why not something like this? On the sustainer curve, display the names of all stages involved in that time (i.e. all stages that haven't separated from the sustainer yet). For instance between t=0s and t=1.2s, all stages are still connected to the sustainer, so display "Sustainer + Booster 1 + Booster 2". From t=1.2s to t=2s, only Booster 1 is still connected, so display "Sustainer + Booster 1" on the sustainer curve, and display "Booster 2" on the booster 2 curve. After t=2s, all stages have separated from the sustainer, so only display "Sustainer" on the sustainer curve.

[image]

neilweinstock, 2024-08-07:
It's hard to argue against that because it's complete unambiguous.  My only fear would be that the tooltip would become a big unwieldy with the multiple names, but it's probably not a real problem.

I say let's go with that.

PR #2530 (2024-08-18): [fixes #2521]
This PR fixes #2521 in which sim plot tooltips did not include the stage name for the main rocket. This included the sustainer stage and any boosters before they separated. 

There were two main problems:
1) When setting the description for the sustainer series, its corresponding branch name was not used, so only the data type and units were passed to the tooltip.
2) The main rocket's branch was modeled as the sustainer stage, but for multi-stage rockets, one or more boosters will be attached to the rocket for a portion of the flight. Therefore, the different portions of the flig [... cut]
commit 29598ef1 (2024-08-18): Merge pull request #2530 from NCoppola25/issue-2521

[fixes #2521]
closed 2024-08-18 by SiboVG