openrocket/openrocket #2541: Strange CP calculation inconsistency

neilweinstock, 2024-08-21:
Look at the CP location on the following two nearly identical files (included in the attached zip file):
[image]
[image]

Although the component analysis shows only very slight differences, the aggregate CP location on the rocket figure is wildly different.

But there's more weirdness:  In the second design, the fins appear to be rotate by a certain amount, even though they're set to rotation of 0.

Collectively, I have no idea what's going on here, but it seems like some sort of bug.

Original report: https://www.rocketryforum.com/threads/water-rocket-cp-calculation-with-open-rocket.187953/

[waterBottleRocketCalcsCp.zip](https://github.com/user-attachments/files/16691864/waterBottleRocketCalcsCp.zip)

SiboVG, 2024-08-21:
I'm not seeing this in the latest unstable.
[image]

SiboVG, 2024-08-21:
> But there's more weirdness: In the second design, the fins appear to be rotate by a certain amount, even though they're set to rotation of 0.

This is normal. It's because you opened the component analysis, which has the wind direction set to 178° (because the "Worst" toggle button is enabled).

closed 2024-08-21 by SiboVG

neilweinstock, 2024-08-21:
> This is normal. It's because you opened the component analysis, which has the wind direction set to 178° (because the "Worst" toggle button is enabled).

Ha, I learned something today.

Was there a known change since last release that explains why this problem went away?

SiboVG, 2024-08-21:
> Was there a known change since last release that explains why this problem went away?

I don't know, maybe @JoePfeiffer does.

JoePfeiffer, 2024-08-21:
> > Was there a known change since last release that explains why this problem went away?
> 
> I don't know, maybe @JoePfeiffer does.

Not specifically... The more I look at the 23.09 results the screwier it gets; in particular, the CP as reported in the design window exactly matches the finset CP as reported in the component analysis for waterBottleRocketCalcsCpTail, but not for waterBottleRocketCalcsCpForward. The total CP as reported in the component analysis (in 23.09) matches the total CP as reported in the design window for current unstable. I can see a few relevant changes in BarrowmanCalculator relating to component assemblies, but nothing jumps out at me as being the one that must have fixed it.