openrocket/openrocket #2543: [Bug] Launch lug pressure CD is extremely high for Mach values very close, but over 1

SiboVG, 2024-08-23:
### Describe the bug

When you enter an extremely small number, that is just over 1 (e.g. 1.0000000000000007), the pressure CD of launch lugs is extremely high.
[image]

Note that this does not happen when you input '1', or a small number just below 1 (e.g. 0.9999999999999993).

### To Reproduce

1. Open 'A simple model rocket'
2. Open the Component Analysis Tool
3. Copy this value to your clipboard: 1.0000000000000007
4. Paste the value in the Mach Number spinner 
5. Extremely high pressure CD

### OpenRocket version

unstable

### What platform are you running on?

macOS

JoePfeiffer, 2024-08-26:
I'm seeing this too.
Cut-and-pasting 1.0000000000000007 into Mach gives really high pressure CD
Change Mach value to .1; pressure CD is reasonable
Typing or cut-and-pasting 1.0 still gives a reasonable CD

it's important to not just go straight from 1.0000000000000007 to 0.9999999999999993; the values are close enough it doesn't get recalculated (which caused me a few minutes of confusion).

SiboVG, 2024-08-26:
> it's important to not just go straight from 1.0000000000000007 to 0.9999999999999993; the values are close enough it doesn't get recalculated (which caused me a few minutes of confusion).

Me too! I didn't know that was the reason for it; I just ignored it :)

JoePfeiffer, 2024-08-26:
> > it's important to not just go straight from 1.0000000000000007 to 0.9999999999999993; the values are close enough it doesn't get recalculated (which caused me a few minutes of confusion).
> 
> Me too! I didn't know that was the reason for it; I just ignored it :)

I was 99% sure we had two bugs before I finally realized what was going on!

SiboVG, 2024-08-28:
The error is caused by the Prandtl-Glauert correction to the friction coefficient in `TubeCalc.java`:
```java
// If we're supersonic, apply a correction
if (conditions.getMach() > 1) {
    f = f / conditions.getBeta();
}
```

Beta is calculated as `sqrt(abs(1 - Mach^2))`, which becomes very small when Mach moves towards 1. As a consequence, the division of f by beta becomes very large.

@JoePfeiffer I'm heading into very unknown territory, but have you considered the Van Driest II method for calculating the pressureCD in `TubeCalc.java`?

JoePfeiffer, 2024-08-29:
> @JoePfeiffer I'm heading into very unknown territory, but have you considered the Van Driest II method for calculating the pressureCD in `TubeCalc.java`?

Pretty unknown for me as well! I was starting to think maybe we could consider having choked flow take over from Darcy-Weissbach when the pressure drop gets too great, but hadn't really explored it yet. I'll look at VanDriest II as well, thanks.

SiboVG, 2024-09-01:
In "Active Guidance and Dynamic Flight Mechanics for Model Rockets", David Ketchledge (1993), it is stated that the correction factor should be clamped to 0.6 between a Mach of 0.8 and 1.1. What do you think @JoePfeiffer? Also: why do we only apply the Prandtl correction at supersonic speeds, I thought it needs to be applied to both subsonic and supersonic?

[image]

PR #2567 (2024-10-02): Remove Prandtl-Glauert correction factor
I can't find any notes giving any clue what I was thinking when I applied the Prandtl-Glauert transformation to the friction coefficient of the tube, and the papers I went from in developing the code also give me no hints. It's obviously completely invalid here. Mea culpa.

Removing it makes no difference to the accuracy of the sim results for the subsonic (B and C motor) test rocket, only slightly reduces it for the supersonic (K motor) tests, and actually improves it slightly (but still not significantly) for the low transonic peak velocity (J) rocket [... cut]
closed 2024-10-02 by JoePfeiffer