openrocket/openrocket #2553: [Bug] Unnecessary "Forward end of airframe is open" warnings in Booster stage of "Three-stage rocket"

SiboVG, 2024-09-01:
### Describe the bug

For some reason, the body tubes in the booster stages in the "Three-stage rocket" example design show 2 warnings that the forward end of the airframe is open. Interestingly enough, those warnings disappear when you disable the sustainer.

### To Reproduce

1. Open the "Three-stage rocket" example design
2. Go to the simulations tab
3. Rerun all sims
4. The sim warnings appear

### OpenRocket version

unstable

### What platform are you running on?

macOS

JoePfeiffer, 2024-09-30:
Question here is under what conditions the warning should appear?

1. Booster has no recovery system (as in the 3 stage example). This will be either a black powder staged rocket with a tumble recovery booster, or a high altitude shot with a ballistic booster recovery. In either case the flyer doesn't care what happens to the booster and the warning shouldn't appear.
2. Booster has a recovery system, and uses motor eject to deploy. The user is likely to be using OR to try to set the motor delay, and should be warned the sim after separation will be inaccurate. But maybe the warning should be specific about calculated optimal delay is likely to be wrong?
3. Booster has a recovery system, and uses electronics to deploy. Results will be inaccurate, but how much is the user likely to care? I could go both ways on this one.

SiboVG, 2024-10-01:
> Question here is under what conditions the warning should appear?
>
> 1. Booster has no recovery system (as in the 3 stage example). This will be either a black powder staged rocket with a tumble [...]

I'm sorry, am I missing something? What does this have to do with the forward airframe being open?

JoePfeiffer, 2024-10-01:
> I'm sorry, am I missing something? What does this have to do with the forward airframe being open?

The airframe is actually open in all these cases. The question is whether the user is likely to be expecting to simulate the booster flight post-separation, in which case they should be reminded that they aren't going to get an accurate result.

JoePfeiffer, 2024-10-02:
Related -- there's no need to flag a "high angle of attach" warning on a booster stage that's starting to tumble (as is the case for both stages in this example).

PR #2594 (2024-11-21): Don't set open airframe warning if rocket is unstable, and clarify wording
This PR identifies rockets (and rocket components) that are aerodynamically unstable, and does not set an open airframe warning in that case. The intent is to avoid spurious open airframe warnings on typical low power boosters after separation. If the booster is stable (as is the case for some high power boosters) the warning is set.

It also clarifies the wording of the warning itself; rather than simply stating that the airframe is open it states that this will case inaccurate altitude predictions.

Here's the warning [... cut]
closed 2024-11-21 by SiboVG