openrocket/openrocket #2564: [Feature Request] Move stage enable/disable inside configuration tab

LithosphereRocketry, 2024-09-26:
**Is your feature request related to a problem? Please describe.**
It feels unintuitive to me that the way to enable or disable a stage per-configuration is in the preview pane, alongside the view type, zoom, CG/CP display, etc. This makes it look like a preview option rather than something that actually changes the simulation configuration.

**Describe the solution you'd like**
I'd like enabling and disabling stages to be done from within the Configuration tab. This would make it align with other per-configuration settings like motor ignition delay and parachute deployment.

**Describe alternatives you've considered**
This could also go in the Simulation tab, but I don't think that makes a lot of sense, as you're likely to want to use different motor configurations for single- and multistage flights. Within the Configuration tab, I can see it making sense in the Motors, Stages, or its own sub-tab; I think any of those could work fine.

**Additional context**
N/A

JoePfeiffer, 2024-11-21:
The most common use of the disabling is watching what happens with stability when various combinations of stages are enabled, so we definitely don't want to lose it from the design window. But playing with it a little, yeah, its relationship to configurations is unclear at best....

LithosphereRocketry, 2024-12-06:
In my head, I was imagining that the staging settings in the design window would just affect display and the calculated stats in that window like mass, CG, stability, etc - I believe this is what it used to do in 15.03.

SiboVG, 2024-12-06:
> In my head, I was imagining that the staging settings in the design window would just affect display and the calculated stats in that window like mass, CG, stability, etc - I believe this is what it used to do in 15.03.

Yes, this was changed in OR 22.02 I believe.

JoePfeiffer, 2024-12-06:
> In my head, I was imagining that the staging settings in the design window would just affect display and the calculated stats in that window like mass, CG, stability, etc - I believe this is what it used to do in 15.03.

We had users who found that behavior really counterintuitive. They expected a disabled stage to be truly disabled.