openrocket/openrocket #2569: Include Loki 76mm motors in 75mm motor filter

neilweinstock, 2024-10-06:
Loki motors are listed as 76mm but apparently fit in the same motor mounts as 75mm motors.  When the motor filter slider in the lower right is set to 75mm only (or, match mount size), the Loki motors should be included.

JoePfeiffer, 2024-10-06:
The issue as stated sounds like a database (ie not our) problem. But I'm surprised, playing with it, that (1) there seem to be unmarked stops in the diameter slider (so I don't seem to be able to set it to anything between 75 and 98 mm), and (2) I can't just enter values like a I can the length -- something you'd think I would have noticed before now...

Oh, this is very interesting: if I pick a LOC 3 in motor mount tube (ID is 76.2mm), the filter is set to 75mm and Loki doesn't fit.

neilweinstock, 2024-10-06:
I initially thought I could go in a do a quick fix to this, but then I decided it was not so easy or obvious.  The best approach I could think of is to apply a +-1 spread to each filter option, which would pull in the 76 mm motors but not affect anything else.  There are some other oddball sized motors in the database but I don't think any of them are mount-compatible with a standard size, and all are separated by more than 1mm.

It's a hack but I suspect a hack is what is needed here.

neilweinstock, 2024-10-06:
We have a similar issue with motor length, such some Aerotech motors who's listed length includes the nozzle and/or ejection cap.  As a result some motors which would fit are excluded by the "Limit motor length to mount length" option.  I'm not sure how much slop we'd need to allow there to fix that problem.

JoePfeiffer, 2024-10-06:
Without thinking about it seriously or looking at the code yet, my initial thoughts are that we are probably a victim of overly clever coding here, and the fix is to go the other direction:

1. Get rid of the "limit to motor mount diameter" checkbox. Having a motor diameter larger than the motor mount makes no physical sense, so don't ever allow it. 
2. Have the diameter sliders go from 0 to the motor mount size instead of whatever the max is now, and let them go smoothly instead of in steps of known motor diameters (or whatever is setting steps now). The default is to have them set at 0 and the motor mount inner diameter.
3. Have text entry boxes so the user can directly enter their preferred min and max.
4. Filter the motor diameter by the values set by the sliders.

Note that this will handle the Loki case, since (checking....) a 75mm motor mount tube actually has a 76.3mm ID (unless, of course, the user manually sets the max diameter to 75).

Since you mention "limit to length" -- exactly opposite the diameter case, I can't think when the "limit to length" check mark ever makes sense. For high power rockets you'll typically have the motor extend out the top of the mount; the distance to the top of the tube coupler in the fin section (minus some room for hardware) is usually but not always what's wanted. For low power, you want either the length of the motor clip or the distance from a thrust block to the motor overhang.  Or if the user likes to tape or friction fit their motors, then it's like the high power case.

neilweinstock, 2024-10-06:
> 1. Get rid of the "limit to motor mount diameter" checkbox. Having a motor diameter larger than the motor mount makes no physical sense, so don't ever allow it.

Please don't.  During design phase I'll often do some what-ifs with a larger motor size, in case I want to change the motor mount size.  It should be possible to try out motors larger than the mount.

> 2. Have the diameter sliders go from 0 to the motor mount size instead of whatever the max is now, and let them go smoothly instead of in steps of known motor diameters (or whatever is setting steps [...]

I'm not sure I understand how this solves anything.  If the slider goes from 0-75, how does the user allow 76?  The user would still need to know that the Loki exception exists, and intentionally set the max value above 75.  They can do this just as well now by setting it to 98.

Letting the slider move in 1mm increments rather than by motor only helps a little, it still requires the user to set the max to 76 to let the Loki motors in.

> Since you mention "limit to length" -- exactly opposite the diameter case, I can't think when the "limit to length" check mark ever makes sense. For high power rockets you'll typically have the [...]

I imagine the HPR folks always uncheck that box.

>For low power, you want either the length of the motor clip or the distance from a thrust block to the motor overhang. Or if the user likes to tape or friction fit their motors, then it's like the high power case.

Not exactly.  To take an 18mm mount as an example: the available length for the motor is 70mm (that's the length of a motor hook.)  All the Estes motors show in the motor list as 70mm.  The 18mm Q-jets, on the other hand will fit fine but their length shows as anywhere up to 81mm, because the extra length is not in the casing.   This has caused confusion to users before.  The length is accurate, but it's not actually the number users typically want to see, which is "how long of a motor mount is required for this motor".  Unfortunately thrustcurve.org does not give us that number.

JoePfeiffer, 2024-10-07:
> Please don't. During design phase I'll often do some what-ifs with a larger motor size, in case I want to change the motor mount size. It should be possible to try out motors larger than the mount.

OK...

> > 2. Have the diameter sliders go from 0 to the motor mount size instead of whatever the max is now, and let them go smoothly instead of in steps of known motor diameters (or whatever is setting steps now). The default is to have them set at 0 and the motor mount inner diameter.
> > 3. Have text entry boxes so the user can directly enter their preferred min and max.
> > 4. Filter the motor diameter by the values set by the sliders.
> > 
> > Note that this will handle the Loki case, since (checking....) a 75mm motor mount tube actually has a 76.3mm ID (unless, of course, the user manually sets the max diameter to 75).
> 
> I'm not sure I understand how this solves anything. If the slider goes from 0-75, how does the user allow 76? The user would still need to know that the Loki exception exists, and intentionally set the max value above 75 (if they're allowed to). They can do this just as well now by setting it to 98.

When the motor selector is first opened, it would be set to 76.3; also, if the user turns off "limit to motor mount" and the diameter is set to anything over that, then it would rest to 76.3.

> Letting the slider move in 1mm increments rather than by motor only helps a little, it still requires the user to set the max to 76 to let the Loki motors in.
But that's the thing -- if the user doesn't do anything, the Loki motors would be accepted.  They'd have to deliberately set it to 75 to exclude them.

> I imagine the HPR folks always uncheck that box.

I certainly do.

> Not exactly. To take an 18mm mount as an example: the available length for the motor is 70mm (that's the length of a motor hook.) All the Estes motors show in the motor list as 70mm. The 18mm [...]

Right, we don't have that number; we don't even have the length of a motor clip.  This is why I don't really see limiting to motor mount as useful for low power either.

JoePfeiffer, 2024-10-18:
It turns out there are good reasons for the slider value to snap to "real" motor diameters.  Looking at the slider, you'll notice it's roughly logarithmic. This is because each diameter is a tick mark with evenly spaced tick marks, and the gap between motor diameters increases as the motors get larger (so the space from 13mm to 18 is the same as between 54 and 75).

A very easy fix would simply be to replace the 75mm spot on the slider with a 76.2mm spot. Do you think this would be OK, or would we get complaints that we have a 76mm diameter instead of a 75mm?

neilweinstock, 2024-10-18:
I don't know if we would get serious *complaints* per se, but I do think it would look a bit ugly.  I'd prefer to make it work invisibly behind-the-scenes.

JoePfeiffer, 2024-10-18:
Take a look at PR #2579 and tell me what you think.

PR #2579 (2024-10-20): Separate motor nominal vs actual diameter
In the motor diameter filter, we now separate a motor's "nominal" and "actual" diameter. The nominal diameter is displayed to the user, while the actual diameter is used for filtering. The intent here is to allow Loki 76mm motors to pass the 75mm filter.

Also adds 32mm motors (Aerotech has at least one listed on their website).

In the following screenshot, note that (in English, anyway) the motor diameter is explicitly stated to be nominal.

[image]

Fixes #2569
closed 2024-10-20 by JoePfeiffer