openrocket/openrocket #2573: open rocket bug report - edited with new information.

19BCK, 2024-10-10:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.  Running simulation when error occured
2.  Restarted the program to see if error persisted
3.  Downloaded latest version of program.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):
bkanemobile@gmail.com

(Do not modify anything below this line.)
---------- Exception stack trace ----------
net.sf.openrocket.gui.main.SwingExceptionHandler$InternalException: An exception occurred during the simulation at net.sf.openrocket.gui.main.SwingExceptionHandler.handleErrorCondition(SwingExceptionHandler.java:120) at net.sf.openrocket.gui.simulation.SimulationRunDialog$InteractiveSimulationWorker.simulationInterrupted(SimulationRunDialog.java:454) at net.sf.openrocket.gui.simulation.SimulationWorker.done(SimulationWorker.java:109) at java.desktop/javax.swing.SwingWorker$5.run(SwingWorker.java:750) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:848) at java.desktop/sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:858) at java.desktop/javax.swing.Timer.fireActionPerformed(Timer.java:311) at java.desktop/javax.swing.Timer$DoPostEvent.run(Timer.java:243) at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:771) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:741) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117) at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:191) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:236) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:234) at java.base/java.security.AccessController.doPrivileged(AccessController.java:318) at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:234) at java.desktop/java.awt.Dialog.show(Dialog.java:1080) at java.desktop/java.awt.Component.show(Component.java:1728) at java.desktop/java.awt.Component.setVisible(Component.java:1675) at java.desktop/java.awt.Window.setVisible(Window.java:1036) at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1016) at net.sf.openrocket.gui.simulation.SimulationRunDialog.runSimulations(SimulationRunDialog.java:217) at net.sf.openrocket.gui.simulation.SimulationEditDialog$6.actionPerformed(SimulationEditDialog.java:289) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389) at java.desktop/java.awt.Component.processEvent(Component.java:6391) at java.desktop/java.awt.Container.processEvent(Container.java:2266) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117) at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:191) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:236) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:234) at java.base/java.security.AccessController.doPrivileged(AccessController.java:318) at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:234) at java.desktop/java.awt.Dialog.show(Dialog.java:1080) at java.desktop/java.awt.Component.show(Component.java:1728) at java.desktop/java.awt.Component.setVisible(Component.java:1675) at java.desktop/java.awt.Window.setVisible(Window.java:1036) at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1016) at net.sf.openrocket.gui.main.SimulationPanel.openDialog(SimulationPanel.java:806) at net.sf.openrocket.gui.main.SimulationPanel.openDialog(SimulationPanel.java:812) at net.sf.openrocket.gui.main.SimulationPanel.openDialog(SimulationPanel.java:820) at net.sf.openrocket.gui.main.SimulationPanel$2.mouseClicked(SimulationPanel.java:250) at java.desktop/java.awt.AWTEventMulticaster.mouseClicked(AWTEventMulticaster.java:278) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6629) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389) at java.desktop/java.awt.Component.processEvent(Component.java:6391) at java.desktop/java.awt.Container.processEvent(Container.java:2266) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4584) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.base/java.security.ProtectionDomain$JavaSec [... cut ...] 136) > at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) > at java.base/java.lang.Thread.run(Thread.java:833)
929 643.125 INFO (SwingExceptionHandler.java:142) Running in EDT, showing dialog
930 643.125 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
931 646.928 USER (SwingExceptionHandler.java:220) User requested sending bug report
932 646.928 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

SiboVG, 2024-10-13:
Hello, thank you for the report. Could you please upload the .ork file that causes this issue? You'll have to ZIP the file first, since GitHub doesn't allow you to upload .ork files directly.

19BCK, 2024-10-14:
Sibo:  Thanks for the reply.  Can i simply attach the ork file here ?

SiboVG, 2024-10-14:
> Sibo: Thanks for the reply. Can i simply attach the ork file here ?
> […](#)

Yes, you can drag-and-drop your ZIPPED .ork file into the comment field on GitHub and post it.

JoePfeiffer, 2024-11-15:
No further info from user

closed 2024-11-15 by JoePfeiffer