openrocket/openrocket #2574: Bug

minsay08, 2024-10-12:
--------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.
2.
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IndexOutOfBoundsException: Index 0 out of bounds for length 0 at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:64) at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:70) at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:266) at java.base/java.util.Objects.checkIndex(Objects.java:361) at java.base/java.util.ArrayList.get(ArrayList.java:427) at net.sf.openrocket.rocketcomponent.RocketComponent.getChild(RocketComponent.java:2087) at net.sf.openrocket.gui.main.BasicFrame.<init>(BasicFrame.java:312) at net.sf.openrocket.gui.main.BasicFrame.open(BasicFrame.java:1364) at net.sf.openrocket.gui.main.BasicFrame.open(BasicFrame.java:1290) at net.sf.openrocket.gui.main.BasicFrame.openAction(BasicFrame.java:1190) at net.sf.openrocket.gui.main.BasicFrame.openAction(BasicFrame.java:1201) at net.sf.openrocket.gui.main.BasicFrame$27.actionPerformed(BasicFrame.java:833) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374) at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:354) at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Actions.actionPerformed(BasicMenuItemUI.java:983) at java.desktop/javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1810) at java.desktop/javax.swing.JComponent.processKeyBinding(JComponent.java:2947) at java.desktop/javax.swing.KeyboardManager.fireBinding(KeyboardManager.java:311) at java.desktop/javax.swing.KeyboardManager.fireKeyboardAction(KeyboardManager.java:253) at java.desktop/javax.swing.JComponent.processKeyBindingsForAllComponents(JComponent.java:3040) at java.desktop/javax.swing.JComponent.processKeyBindings(JComponent.java:3032) at java.desktop/javax.swing.JComponent.processKeyEvent(JComponent.java:2909) at java.desktop/java.awt.Component.processEvent(Component.java:6403) at java.desktop/java.awt.Container.processEvent(Container.java:2266) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1952) at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:883) at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1150) at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:1020) at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:848) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4882) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 23.09
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
User-defined thrust curves location: C:\Users\Victor\AppData\Roaming\OpenRocket\ThrustCurves
LAF: com.sun.java.swing.plaf.windows.WindowsLookAndFeel
JOGL version: 2.5.0
Current default locale: pt_BR
System properties:
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar
java.class.version=61.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\Victor\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Common Files\Intel\Shared Files\cpp\bin\Intel64;C:\Program Files\Eclipse Adoptium\jre-8.0.362.9-hotspot\bin;C:\Program Files\Eclipse Adoptium\jre-17.0.6.10-hotspot\bin;C:\Program Files\Eclipse Adoptium\jre-11.0.18.10-hotspot\bin;C:\Program Files (x86)\Eclipse Adoptium\jre-8.0.362.9-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\dotnet\;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\Polyspace\R2019a\bin;C:\Users\Victor\AppData\Local\Programs\Python\Python312\Scripts\;C:\Users\Victor\AppData\Local\Programs\Python\Python312\;C:\Users\Victor\AppData\Local\Programs\Python\Launcher\;C:\Users\Victor\AppData\Local\Microsoft\WindowsApps;
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=17.0.7+7-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=17
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=17.0.7
java.version.date=2023-04-18
java.vm.compressedOopsMode=32-bit
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=17
java.vm.vendor=BellSoft
java.vm.version=17.0.7+7-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln10684334058982385452
line.separator=\u000d\u000a
native.encoding=Cp1252
[... 192 lines cut ...]
331 775.516 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
[... cut ...]
350 775.612 VBOSE (AbstractScaleFigure.java:209) ____ Updating RocketFigure to: 0,280458 user scale, 1060,00 overall scale
351 775.613 VBOSE (AbstractScaleFigure.java:209) ____ Updating RocketFigure to: 0,280458 user scale, 1060,00 overall scale
352 775.614 VBOSE (AbstractScaleFigure.java:209) ____ Updating RocketFigure to: 0,280458 user scale, 1060,00 overall scale
353 782.339 USER (SwingExceptionHandler.java:220) User requested sending bug report
354 782.358 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2024-10-15:
Thank you for your report. It appears as if it may be resulting from an error in your .ork file -- if possible, please zip it (github won't let us upload a .ork file directly) and add it to your report.

JoePfeiffer, 2024-10-25:
No further info from user

JoePfeiffer, 2024-11-15:
No further info

closed 2024-11-15 by JoePfeiffer