openrocket/openrocket #2583: [Bug] Missing Wayland support

sheepymeh, 2024-10-21:
### Describe the bug

OpenRocket does not support Wayland on Linux. Launching gives the error:

```
java -Dopenrocket.debug -jar /usr/share/java/openrocket/openrocket.jar
New classpath:
   file:/usr/share/java/openrocket/openrocket.jar
   classpath:lib/gluegen-rt.jar
   classpath:lib/jogl-all.jar
111            INFO  [main] n.s.o.s.SwingStartup - Starting up OpenRocket version 23.09
123            INFO  [main] n.s.o.s.SwingStartup - Checking for graphics head
129            INFO  [main] n.s.o.s.SwingStartup - Moving startup to EDT
179      StdErr DEBUG [main] STDERR - Exception in thread "main" java.lang.RuntimeException: Error starting OpenRocket
179      StdErr TRACE [main] STDERR - 	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:40)
179      StdErr TRACE [main] STDERR - 	at net.sf.openrocket.startup.OpenRocket.main(OpenRocket.java:36)
180      StdErr DEBUG [main] STDERR - Caused by: java.lang.reflect.InvocationTargetException
180      StdErr TRACE [main] STDERR - 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
180      StdErr TRACE [main] STDERR - 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
180      StdErr TRACE [main] STDERR - 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
180      StdErr TRACE [main] STDERR - 	at java.base/java.lang.reflect.Method.invoke(Method.java:569)
180      StdErr TRACE [main] STDERR - 	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:38)
180      StdErr TRACE [main] STDERR - 	... 1 more
180      StdErr DEBUG [main] STDERR - Caused by: java.awt.AWTError: Can't connect to X11 window server using ':0' as the value of the DISPLAY variable.
181      StdErr TRACE [main] STDERR - 	at java.desktop/sun.awt.X11GraphicsEnvironment.initDisplay(Native Method)
181      StdErr TRACE [main] STDERR - 	at java.desktop/sun.awt.X11GraphicsEnvironment$1.run(X11GraphicsEnvironment.java:105)
181      StdErr TRACE [main] STDERR - 	at java.base/java.security.AccessController.doPrivileged(AccessController.java:318)
181      StdErr TRACE [main] STDERR - 	at java.desktop/sun.awt.X11GraphicsEnvironment.initStatic(X11GraphicsEnvironment.java:64)
181      StdErr TRACE [main] STDERR - 	at java.desktop/sun.awt.X11GraphicsEnvironment.<clinit>(X11GraphicsEnvironment.java:59)
181      StdErr TRACE [main] STDERR - 	at java.desktop/sun.awt.PlatformGraphicsInfo.createGE(PlatformGraphicsInfo.java:36)
181      StdErr TRACE [main] STDERR - 	at java.desktop/java.awt.GraphicsEnvironment$LocalGE.createGE(GraphicsEnvironment.java:91)
181      StdErr TRACE [main] STDERR - 	at java.desktop/java.awt.GraphicsEnvironment$LocalGE.<clinit>(GraphicsEnvironment.java:82)
182      StdErr TRACE [main] STDERR - 	at java.desktop/java.awt.GraphicsEnvironment.getLocalGraphicsEnvironment(GraphicsEnvironment.java:104)
182      StdErr TRACE [main] STDERR - 	at java.desktop/sun.awt.X11.XToolkit.<clinit>(XToolkit.java:224)
182      StdErr TRACE [main] STDERR - 	at java.desktop/sun.awt.PlatformGraphicsInfo.createToolkit(PlatformGraphicsInfo.java:40)
182      StdErr TRACE [main] STDERR - 	at java.desktop/java.awt.Toolkit.getDefaultToolkit(Toolkit.java:599)
182      StdErr TRACE [main] STDERR - 	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1346)
182      StdErr TRACE [main] STDERR - 	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1480)
182      StdErr TRACE [main] STDERR - 	at net.sf.openrocket.startup.SwingStartup.main(SwingStartup.java:89)
183      StdErr TRACE [main] STDERR - 	... 6 more```

### To Reproduce

1. Start a Wayland-only session with XWayland disabled
2. Try to start OpenRocket

### Screenshots / .ork file

NIL

### OpenRocket version

23.09

### What platform are you running on?

Linux

### Additional context

I am running Sway on Arch Linux. The application works fine with XWayland enabled.

closed 2024-10-21 by JoePfeiffer

sheepymeh, 2024-10-21:
I see, thank you for letting me know. It seems like JetBrains has managed to [implement the Wayland toolkit](https://blog.jetbrains.com/platform/2024/07/wayland-support-preview-in-2024-2/) in their IDEs. Would upgrading Java versions be necessary for this change in OpenRocket?

JoePfeiffer, 2024-10-22:
Talking to the other developers, it looks like IntelliJ is able to use Wayland for its own UI, but not in programs built using it (I'll be happy to be corrected!). Though that must mean it's getting closer...

sheepymeh, 2024-10-22:
I see, thanks a lot and fingers crossed for better Wayland support in Java soon!