openrocket/openrocket #2584: [Bug] Material search test sometimes fails in unit tests

JoePfeiffer, 2024-10-21:
### Describe the bug

Under conditions I can't reliably duplicate, the materials database search sometimes returns materials named <[material:Aluminum]> instead of <Aluminum> which causes unit tests to fail.  When unit tests are re-run, it succeeds.  It does seem like every time one of these tests fails they both do.  Cutting and pasting from the unit test results:
testFindMaterialByTypeAndName()

org.opentest4j.AssertionFailedError: expected: <Aluminum> but was: <[material:Aluminum]>
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertionFailureBuilder.build(AssertionFailureBuilder.java:151)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertionFailureBuilder.buildAndThrow(AssertionFailureBuilder.java:132)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertEquals.failNotEqual(AssertEquals.java:197)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertEquals.assertEquals(AssertEquals.java:182)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertEquals.assertEquals(AssertEquals.java:177)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.Assertions.assertEquals(Assertions.java:1145)
	at app/info.openrocket.core@23.09.SNAPSHOT/info.openrocket.core.database.MaterialDatabaseTest.testFindMaterialByTypeAndName(MaterialDatabaseTest.java:61)
	at java.base@17.0.13-ea/java.lang.reflect.Method.invoke(Method.java:569)
	at java.base@17.0.13-ea/java.util.ArrayList.forEach(ArrayList.java:1511)
	at java.base@17.0.13-ea/java.util.ArrayList.forEach(ArrayList.java:1511)

testFindMaterialByTypeNameAndDensity()

org.opentest4j.AssertionFailedError: expected: <CustomMaterial> but was: <[material:CustomMaterial]>
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertionFailureBuilder.build(AssertionFailureBuilder.java:151)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertionFailureBuilder.buildAndThrow(AssertionFailureBuilder.java:132)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertEquals.failNotEqual(AssertEquals.java:197)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertEquals.assertEquals(AssertEquals.java:182)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.AssertEquals.assertEquals(AssertEquals.java:177)
	at app/org.junit.jupiter.api@5.10.2/org.junit.jupiter.api.Assertions.assertEquals(Assertions.java:1145)
	at app/info.openrocket.core@23.09.SNAPSHOT/info.openrocket.core.database.MaterialDatabaseTest.testFindMaterialByTypeNameAndDensity(MaterialDatabaseTest.java:70)
	at java.base@17.0.13-ea/java.lang.reflect.Method.invoke(Method.java:569)
	at java.base@17.0.13-ea/java.util.ArrayList.forEach(ArrayList.java:1511)
	at java.base@17.0.13-ea/java.util.ArrayList.forEach(ArrayList.java:1511)

### To Reproduce

1. 
2. 
3. 
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

current unstable (not sure whether this is a regression)

### What platform are you running on?

Linux

SiboVG, 2024-11-21:
Looks like a race condition with the translator injection for the unit tests.

SiboVG, 2024-12-07:
@JoePfeiffer did you ever encounter this unit test failure on your local machine? I just ran 300,000 iterations of the `testName` test on my local machine (macOS) and did not get a single failure.

JoePfeiffer, 2024-12-07:
> @JoePfeiffer did you ever encounter this unit test failure on your local machine? I just ran 300,000 iterations of the `testName` test on my local machine (macOS) and did not get a single failure.

Yes, both of my local Linux machines. How do you the tests multiple times? I can try to get a sense of just how frequent it is.

SiboVG, 2024-12-07:
> Yes, both of my local Linux machines.

Interesting. I'll try to run the unit tests on my Linux boot.

> How do you the tests multiple times? I can try to get a sense of just how frequent it is.

Created a new test method in the same file, added a for-loop with 100,000 iterations with the `testName` method inside, and ran that test.

SiboVG, 2024-12-07:
Okay, interesting, so I can recreate this on my Linux boot! So it's platform-specific...

SiboVG, 2024-12-07:
The substitution also doesn't work when running OR.

[image]

JoePfeiffer, 2024-12-07:
Is that running unstable or this broken PR? Substitution was working for me until I made the attempted fix (which I've now reverted).

SiboVG, 2024-12-08:
> Is that running unstable or this broken PR? Substitution was working for me until I made the attempted fix (which I've now reverted).

That was on #2605. Tested on unstable and the UI is normal again. Haven't tested the unit testing yet.

SiboVG, 2024-12-08:
Oh okay, the unit tests from #2605 also fail on macOS.

closed 2025-10-23 by SiboVG