openrocket/openrocket #2592: [Feature Request] make plot legend sensitive and enbolden plot lines

JoePfeiffer, 2024-11-10:
**Is your feature request related to a problem? Please describe.**
When a user is examining simulation results, the various types of flight data being displayed are color coded with a legend at the bottom of the window identifying each color. It is sometimes (particularly when many types of data are being displayed) difficult to determine which plot line is actually being displayed in which color.

**Describe the solution you'd like**
If the user hovers the mouse over one of the plot types in the legend, make that plot line bold so it stands out.

**Describe alternatives you've considered**
First, of course, one can already hover over a data point; the tooltip displays which data type is being viewed. This seems sort of backward, as I end up hovering over several lines trying to find the one I'm looking for.

Also, we could display the color swatch in the legend as a small block instead of a line; the greater area will make the color of the swatch clearer. This might be a good idea in addition to the enhancement being requested here.

neilweinstock, 2025-02-03:
This is a good idea.

SiboVG, 2025-11-18:
> Hello JoePfeiffer,
> 
> I'm Heitor. As part of a school project, I would like to work on this issue.

Hi Heitor, cool! Sure, you can work on this. Thanks!

neilweinstock, 2025-11-18:
That sounds good.

closed 2025-12-05 by JoePfeiffer