openrocket/openrocket #2595: [Bug] Simming only booster stages gives impossible warnings

JoePfeiffer, 2024-11-18:
### Describe the bug

If the sustainer of a multistage rocket is disabled and the rocket is simulated, impossible warnings are generated. Here's the result of simming the three stage example, with the sustainer disabled:
[image]
The first two warnings in this tooltip are impossible: there is no nose cone, and there are no pods.

### To Reproduce

1. Launch OR 23.09
2. Open the "Three-stage rocket" example.
3. Above the rocket figure window, click "Sustainer" to disable it. At this point the first and second stages should be visible.
4. Run a simulation.
5. Hover over the red explanation point to bring up the tooltip.
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

23.09 (still present in current unstable)

### What platform are you running on?

Linux

### Additional context

If this were a regression, I'd call it a beta blocker, but it's in 23.09, and has not been reported before now.

SiboVG, 2024-11-21:
See [this comment](https://github.com/openrocket/openrocket/pull/2594#issuecomment-2492286325): for a "Three stage rocket" example, when you disable the sustainer from the 2nd flight config, the 1st and 3rd simulation somehow get the warnings that you would expect for sim 2.

SiboVG, 2024-12-18:
This is weird, looks like the simulations are affected by the **currently selected flight configuration**.

Selected the flight config with the sustainer disabled (note that this should only affect sim 2, sim 1 and 3 still have the sustainer enabled):
[image]

Selected the flight config with the sustainer enabled:
[image]

JoePfeiffer, 2024-12-18:
Thinking about #2564 here -- our enable/disable flags are on the currently selected flight configuration, which typically won't be the flight configuration being simulated.

SiboVG, 2024-12-18:
I've traced the bug back to this code in `setAfter()` of `RocketComponent.java`:

```java
// If previous components are inactive, set this as the new reference point
if (!getRocket().getSelectedConfiguration().isComponentActive(referenceComponent)) {
    this.position = this.position.setX(0.0);
    return;
}
```

Now I just have to find a way to fix it :)

SiboVG, 2025-01-03:
So the issue is that the `position` and `axialOffset` in `RocketComponent` are calculated using the currently selected flight configuration. An example:

1. In a "Three-stage rocket" example, if I disable the upper `Booster stage` in the currently selected flight config (`C6-5; B6-0; B6-0` becomes `C6-5; ; B6-0`), the `position` and `axialOffset` are recalculated.
2. If I rerun Simulation 1 (with config `A8-5; B6-0; B6-0`), it uses the cached `position` and `axialOffset` of each component. However, those were calculated in config `C6-5; ; B6-0` and do not reflect the state of config `A8-5; B6-0; B6-0`.

I see two solutions to this problem:
1. When running Simulation 1, first recalculate the `position` and `axialOffset` with the flight configuration for that simulation (`A8-5; B6-0; B6-0`). After simulation 1 is finished running, recalculate `position` and `axialOffset` for the currently selected flight config (`C6-5; ; B6-0`), so we're back at the original state.
2. Change the cached `double position` and `double axialOffset` variables to `Map<FlightConfigurationId, Double> positionMap` and `Map<FlightConfigurationId, Double> axialOffsetMap`. So, you always calculate the `position` and `axialOffset` for each flight configuration, and fetch the correct value when needed.

Option 1 is the easiest, but feels a bit like a hack. Option 2 is IMO a good long-term solution, but requires a large code rewrite.

senft-research, 2025-05-09:
What is the current state of this one? Is there a branch working on this? If not I'll happily give it a crack? (Wanna focus on bug squashing as my main contribution effort outside my current PR)

SiboVG, 2025-05-09:
> What is the current state of this one? Is there a branch working on this? If not I'll happily give it a crack? (Wanna focus on bug squashing as my main contribution effort outside my current PR)

I've not worked on this for a while and currently don't have the time to do so, so you're very welcome to try it out!

senft-research, 2025-05-09:
I am thinking of trying a slightly different approach (closer to your 2nd suggestion, but something that might drastically reduce the refactoring). Would you be able to let me know roughly where the simulations are calling the values upon running a sim?

senft-research, 2025-05-09:
Update: 

After looking at this more, I think that option 2 is still not quite getting to the core root of the problem. 

In fact, I think the root cause is one that is shared by #2777 . This core issue being the way the data structure of OpenRocket currently is handled, means that data dependent on each other are "independently mutable".

Warnings with UUIDs that no longer exist, Flight Configs being utilized that are out of sync with currently cached positional values / offsets. The data that should be inherently linked to each other is detached in ways that can easily lead to bugs, even after we address this ticket and #2777. 

As someone fresh to the code base, I'd regret not at least voicing the idea that maybe this is a more general issue that should be addressed as a more centralized effort? More specifically, actually doing the monotonous task of mapping out how the data is handled in a diagram, to visually see where things that should be dependent are being independently changed. 

To summarize, I think this issue (and the others related to this issue) might be worth addressing as a larger, more encompassing effort. 

Apologies if I have overstepped with this view. Regardless, your thoughts on this matter would be highly appreciated.