openrocket/openrocket #2596: [Bug] Multi-edit of coefficient of drag doesn't display correctly in component tree

jimmiedave, 2024-11-18:
### Describe the bug

On a model with a lot of external fru-fru, I selected a lot of said fru-fru and overrode the CD to 0.
The component tree did not appear to reflect the override (though see screenshot with screen trash), but upon inspecting individual fru-fru items and closing dialog, correct CD override markings appeared for the inspected fru-fru in the component tree (but **not** for the items that were multi-edited but not inspected individually).

Save and reload of the file shows the overrides correctly, I believe. Haven't tested a bunch.

### To Reproduce

1. Select a bunch of stuff
2. Override its CD to 0
3. Save the override
4. Look at part hierarchy tree
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*
[image]

Note that Centering Ring 3 has correct marking for override. Centering Ring 2 (also overridden) has no marking, but screen trash indicating incorrectly laid-out marking.

#### .ork file:
*(drag-and-drop the file here as a .zip file)*
[3in estes black hole space probe.ork.zip](https://github.com/user-attachments/files/17794577/3in.estes.black.hole.space.probe.ork.zip)

.ork file has 2, 3, 4 overridden. When you load this, you'll probably see these overrides. Save and load of the file appears to fix the formatting issue.

### OpenRocket version

23.09

### What platform are you running on?

macOS

SiboVG, 2024-11-18:
Confirmed on my machine (macOS). Works fine for CG and mass override, not for CD.

jimmiedave, 2024-11-18:
And on Ubuntu 24.04.1 LTS, AMD Ryzen 3400G (integrated graphics)

[image]

Note that here the first selection is shown correctly.

Also, opening/closing the hierarchy group does not restore correct display. Removing CD correctly removes correctly-marked and screen-trash versions.

SiboVG, 2025-09-20:
> Me and another developer have looked into this issue, and it seems to be resolved in the latest version? Can you please confirm?

It does appear to be fixed now, maybe thanks to our switch to the FlatLaf UI.

closed 2025-09-20 by SiboVG