openrocket/openrocket #2597: [Bug] Build fails on latest JDK

DarkTyger, 2024-11-20:
### Describe the bug

NoseConeTest.java:11: error: extraneous semicolon
import static org.junit.jupiter.api.Assertions.assertSame;;

### To Reproduce

1. git clone https://github.com/openrocket/openrocket
1. cd openrocket
1. gradle clean build

Expected
Builds.

Actual
Compile failure.

### OpenRocket version

main

### What platform are you running on?

Linux

### Additional context

Looks like a `;;` is messing with the  build.

JoePfeiffer, 2024-11-21:
Changed it to a feature request, since we're still on 17.

DarkTyger, 2024-11-21:
> Changed it to a feature request, since we're still on 17.

It's a typo, should be a quick fix. Delete the semicolon.

closed 2024-11-21 by SiboVG