openrocket/openrocket #2600: ORBrake plugin errors in OpenRocket simulation

TristanBurchett, 2024-11-23:
**---------- Bug report ----------**

1. I installed the ORBrake plugin for simulations from here: https://github.com/WPI-HPRC/ORBrake?tab=readme-ov-file
2. The extension appeared under simulation options in "edit simulation".
3. However, when I tried to run it, the attached error trace occurred.

[ORBRAKE BUG REPORT.txt](https://github.com/user-attachments/files/17878110/ORBRAKE.BUG.REPORT.txt)

email tburchett.2026@saintannsny.org or tristan.burchett@gmail.com if any progress is made on this bug (exception trace attached).

SiboVG, 2024-11-23:
Hello, thank you for your report. The ORBrake plugin is not maintained by the OpenRocket organization. I see you've already communicated this in the ORBrake repo. in January of this year, I already opened a Pull Request to support the plugin in OR 23.09, see [this PR](https://github.com/WPI-HPRC/ORBrake/pull/5). The PR contains a link to a JAR file. Just download that JAR file and use it as a replacement for the official JAR.

closed 2024-11-23 by SiboVG

rickhg12hs, 2024-11-23:
I notice from the `ORBrake` `README`:

> Please direct inquiries about this program to the HPRC Officer Board at hprc-officers@wpi.edu.

Perhaps a ping there might revive some repo activity.

SiboVG, 2024-11-25:
> I notice from the `ORBrake` `README`:
> 
> > Please direct inquiries about this program to the HPRC Officer Board at [hprc-officers@wpi.edu](mailto:hprc-officers@wpi.edu).
> 
> Perhaps a ping there might revive some repo activity.

Tried it, but I cannot send mails to them because my mail hostname does not match theirs.