openrocket/openrocket #2607: [Feature Request] Add checkbox in plot to toggle visibility of flight events

SiboVG, 2024-12-06:
I believe it could be useful to toggle the visibility of the flight event markers in the sim plots to reduce visual clutter when you don't need the events. You can already choose to not plot any flight event by clicking the "None" button in the plot config dialog, but then you would have to recheck all the events if you do want them plotted.

[image]

AbdallahZerfaoui, 2025-01-25:
Hello Sibo, 

I’ve added a toggle button to quickly show or hide the flight event markers in the simulation plots, as per your suggestion to reduce visual clutter without needing to recheck all the events. Currently, I’ve placed it in the "Edit Simulation" window. However, I’m wondering if it might be more intuitive to position it next to the "Show Data Points" checkbox in the main simulation interface.

Do you think it would be better to keep it in the "Edit Simulation" window or move it to the main plot interface for quicker access?

Thanks,

[image]

SiboVG, 2025-01-25:
Hello @AbdallahZerfaoui, thank you for working on this! I would move it to the simulation plot instead of the simulation edit window. Additionally, I would use a "Show flight events" checkbox instead of a toggle button.

Thanks!

SiboVG, 2025-10-15:
> Hello Sigo,
> 
> I’m Gustave-M. Along with two classmates, we’re working on this issue as part of a school project. We will add a "Show Events" checkbox to the plot, as your example. We hope this issue is still open
> 
> Best regards

Hi Gustave, yes please, go ahead!