openrocket/openrocket #2611: Exception exporting a file to .obj

jimmiedave, 2024-12-10:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.Exporting interceptor.ork as interceptor.obj for 3D rendering
2.
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IllegalArgumentException: At least 3 points are required to create a polygon
at net.sf.openrocket.file.wavefrontobj.export.shapes.PolygonExporter.verifyPoints(PolygonExporter.java:127)
at net.sf.openrocket.file.wavefrontobj.export.shapes.PolygonExporter.addPolygonMesh(PolygonExporter.java:22)
at net.sf.openrocket.file.wavefrontobj.export.components.FinSetExporter.generateMesh(FinSetExporter.java:57)
at net.sf.openrocket.file.wavefrontobj.export.components.FinSetExporter.addToObj(FinSetExporter.java:46)
at net.sf.openrocket.file.wavefrontobj.export.OBJExporterFactory.handleComponent(OBJExporterFactory.java:263)
at net.sf.openrocket.file.wavefrontobj.export.OBJExporterFactory.doExport(OBJExporterFactory.java:158)
at net.sf.openrocket.gui.main.BasicFrame.saveWavefrontOBJFile(BasicFrame.java:1684)
at net.sf.openrocket.gui.main.BasicFrame.saveAsWavefrontOBJ(BasicFrame.java:1671)
at net.sf.openrocket.gui.main.BasicFrame.exportWavefrontOBJAction(BasicFrame.java:1664)
at net.sf.openrocket.gui.main.BasicFrame$8.actionPerformed(BasicFrame.java:452)
at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374)
at java.desktop/com.apple.laf.ScreenMenuItem.actionPerformed(ScreenMenuItem.java:129)
at java.desktop/java.awt.MenuItem.processActionEvent(MenuItem.java:692)
at java.desktop/java.awt.MenuItem.processEvent(MenuItem.java:651)
at java.desktop/java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:379)
at java.desktop/java.awt.MenuComponent.dispatchEvent(MenuComponent.java:367)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:776)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 23.09
OpenRocket source: default
OpenRocket location: /Applications/OpenRocket23-09.app/Contents/Resources/app/OpenRocket.jar
User-defined thrust curves location: /Users/theowner/Library/Application Support/OpenRocket/ThrustCurves
LAF: com.apple.laf.AquaLookAndFeel
JOGL version: 2.5.0
Current default locale: en_US
System properties:
apple.laf.useScreenMenuBar=true
com.apple.mrj.application.apple.menu.about.name=OpenRocket
exe4j.moduleName=/Applications/OpenRocket23-09.app
file.encoding=UTF-8
[... 245 lines cut ...]
324 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:379)
325 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.MenuComponent.dispatchEvent(MenuComponent.java:367)
326 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:776)
327 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
328 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
329 82.166 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
330 82.166 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
331 82.166 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
332 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
333 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
334 82.166 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
335 82.166 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
336 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
337 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
338 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
339 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
340 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
341 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
342 82.166 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
343 82.166 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
344 82.166 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
345 82.226 WARN (DoubleModel.java:1006) DoubleModel[OBJExportOptions:ScalingDouble] being garbage-collected while having listeners [javax.swing.JSpinner$ModelListener@2bbb0041, net.sf.openrocket.file.wavefrontobj.OBJOptionChooser$10@78e1fc48, net.sf.openrocket.file.wavefrontobj.OBJOptionChooser$7@688b0f8d]
346 88.791 USER (SwingExceptionHandler.java:220) User requested sending bug report
347 88.796 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

jimmiedave, 2024-12-10:
This was with OR 23.09 (NOT beta).

[interceptor.ork.zip](https://github.com/user-attachments/files/18084682/interceptor.ork.zip)

jimmiedave, 2024-12-10:
[image]

SiboVG, 2024-12-10:
The issue is in the `front`component in the `Top and front flat surfaces`, which is just a line (hence the descriptive error message).

[image]

We should probably just ignore polygons with a smaller number of polygons than 3 and add it as a warning.

SiboVG, 2024-12-10:
(confirmed on unstable btw)

SiboVG, 2024-12-10:
Actually, disregard my earlier suggestion of ignoring the faulty component and adding a warning message. That's not a clean solution because the `front` component, being just a line, should never have been exported in the first place. Instead, we should implement proper OBJ export error messages in addition to the existing warning messages.

When implementing this, I would create a custom Exception class for OBJ exporting, then catch that exception somewhere, and add it as an error.

Maybe we can also add a checkbox in the OBJ export error-warning dialog to ignore components with errors.

This would be a better long-term solution, but it means this issue will not be fixed by the 24.12 release.

jimmiedave, 2024-12-10:
Since the model was @neilweinstock 's work-in-prog, not sure that it would be a problem on his finished model. Was grabbing the closest model file at hand with decals to try something out.

Your discussion points make sense. Probably a rare-enough event that ignoring it for 24.12 inconveniences nobody.

PR #2890 (2025-09-20): Account for zero-thickness and zero-length components in OBJ export
This PR fixes #2881, fixes #2794, and fixes #2611. OBJ exporting now better accounts for components that have a 0 thickness or length.