openrocket/openrocket #2613: HTML escaping problem in simulation warnings panel

neilweinstock, 2024-12-15:
I think this problem was previously found and fixed elsewhere (maybe the warnings in the Rocket Figure display?)

[image]

SiboVG, 2024-12-15:
Can you share the .ork file for this? I'm not seeing this on the latest unstable, are you running the latest unstable? I also wonder why your warnings are different from mine. If I change the body tube diameter in A simple model rocket, the warning first shows the components and then the warning message, whereas it's the reverse for you.

[image]

neilweinstock, 2024-12-15:
Huh, weird.

This is with the recent beta packaged installers Joe made, but I also see it with the current up-to-date builds I'm fiddling with.

Here's the ORK.  Pretty sure I created this with the official 23.09, but I can't be 100% sure (unless it's written into the XML file somewhere).
[boostershot.ork.zip](https://github.com/user-attachments/files/18141184/boostershot.ork.zip)

SiboVG, 2024-12-15:
If you re-run the sims in the latest unstable version, the formatting is correct (warning texts get updated when you re-run the sims). Should we close this issue then?
[image]

neilweinstock, 2024-12-15:
Rerunning the simulations on the latest build fixes the problem.  Closing.

closed 2024-12-15 by neilweinstock