openrocket/openrocket #2614: [Bug] Incorrect default nose cone weights

BarryF-2024, 2024-12-16:
### Describe the bug

There are a few nose cones with default weights in Open Rocket that are quite a bit higher than what I am finding as I build my rockets.

All weights below are in ounces with the tab removed from the shock cord anchor point- if there was one. I weighed as many cones as I had access to since the weights can vary.

Part number | Open Rocket default weight | # of nose cones weighed and their weights (rocket associated with nose cone)

Estes 72413 2.5” – OR 15.6 - weighed 3: 2.01, 2.45 and 2.5
Estes 72653 2” – OR 9.79 – weighed 4: 1.32, 1.60, 1.63 and 1.65 
Balsa Machining PNC300 – OR 25.3 - weighed 3: 3.7, 4.25 and 4.3
Estes 72695 3” – OR 18.5 – weighed 1 @ 2.80 (Big Daddy/Levithan)
Estes 72753 3” – OR 24.9 - weighed 2: 3.84 and 4.53 (Nike Smoke)

### To Reproduce

1. Add any of those nose cones to your sim
2. Note the default weights
3. 
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

23.09

### What platform are you running on?

macOS

### Additional context

I hope this helps, and these are all I have found thus far. I know this is not a bug, but I wasn't sure how else to classify this issue. These weights have been like this in Open Rocket for around 2 years of releases.

SiboVG, 2024-12-16:
Interesting... I think the issue is the wall thickness for those components. For instance, the Estes 72413 is set to default `filled`with a wall thickness of `(base diameter) / 2`. If you uncheck the filled checkbox and reduce the wall thickness to 0.2 cm, you get a weight of 66.5 g (2.35 oz).

I imagine that the real-life nose cones are hollow on the inside? When you check the `Filled` checkbox, OR assumes the object is solid, which is not the case for every part.

JoePfeiffer, 2024-12-16:
The bad mass turns out to be a symptom of the actual database error.  Looking at the 72413, see a couple of the parameters:

        <Length Unit="in">9.75</Length>
        <Thickness>0.125</Thickness>
        
So the thickness is 0.125 meters instead of inches. I expect we'll find the others in the report are similar.

Thank you!

SiboVG, 2024-12-16:
Ah, see #2480.

JoePfeiffer, 2024-12-16:
Ouch, I even made the same observation then and have no recollection of it.  Let's make *sure* we fix it before final release this time.

SiboVG, 2024-12-16:
> The bad mass turns out to be a symptom of the actual database error. Looking at the 72413, see a couple of the parameters:
>
> ```
>     <Length Unit="in">9.75</Length>
>     <Thickness>0.125</Thickness> [...]

That's true for Estes 72413 and Estes 72653, but all the other parts already had their thickness unit set correctly. However, they have the `Filled` checkbox checked. If you uncheck that, you see that it toggles to the correct wall thickness. Is there something else setting the nose cone to filled?

SiboVG, 2024-12-16:
Woooooow, okaaaay, you just found a big issue @BarryF-2024! In the nose cone preset loader, the filled setting is overwritten by `loadFromPreset` in `SymmetricComponent`:
```Java
@Override
protected void loadFromPreset(ComponentPreset preset) {
if (preset.has(ComponentPreset.THICKNESS)) {
	this.thickness = preset.get(ComponentPreset.THICKNESS);
	this.filled = false;
}
if (preset.has(ComponentPreset.FILLED)) {
	this.filled = true;
}

super.loadFromPreset(preset);
}
```

This is a big oopsie...

BarryF-2024, 2024-12-17:
Wow @SiboVG, I am glad I could help out. I am a user as the coding side is not anything I dabble in. Thanks to Joe Pfeiffer on Facebook who suggested I post this here.

closed 2024-12-18 by JoePfeiffer