openrocket/openrocket #2622: [Bug] Can't initialize Photo Studio on Windows 11 ARM (tested Parallels VM on macOS)

jimmiedave, 2024-12-20:
### Describe the bug

Exception when opening Photo Studio from the menu:

java.lang.NullPointerException: Cannot invoke "java.awt.Component.addMouseWheelListener(java.awt.event.MouseWheelListener)" because "this.canvas" is null at info.openrocket.swing.gui.figure3d.photo.PhotoPanel.setupMouseListeners(PhotoPanel.java:234) at info.openrocket.swing.gui.figure3d.photo.PhotoPanel.<init>(PhotoPanel.java:144) at info.openrocket.swing.gui.figure3d.photo.PhotoFrame.<init>(PhotoFrame.java:97) at info.openrocket.swing.gui.figure3d.photo.PhotoFrame.<init>(PhotoFrame.java:71) at info.openrocket.swing.gui.main.BasicFrame$19.actionPerformed(BasicFrame.java:705) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374) at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1028) at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1072) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389) at java.desktop/java.awt.Component.processEvent(Component.java:6391) at java.desktop/java.awt.Container.processEvent(Container.java:2266) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:775) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:747) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:744) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

### To Reproduce

1. Open a known-good model in 24.12 b01 Windows ARM on a Parallels ARM VM.
2. Try to start photo studio from the menu
3. Get Exception
...

### Screenshots / .ork file

#### Screenshot(s):
[image]

#### .ork file:
[BETA_Daves38mmScreechAsBuilt.ork.zip](https://github.com/user-attachments/files/18214984/BETA_Daves38mmScreechAsBuilt.ork.zip)

### OpenRocket version

24.12 beta .01

### What platform are you running on?

Windows

jimmiedave, 2024-12-20:
Same model, same action, same hardware machine (native macOS m3 MAX) works fine in macOS OpenRocket 23.09 SNAPSHOT (don't have 24.01 beta for Apple Silicon yet).

jimmiedave, 2024-12-20:
Also can't view "3D Finished" or other 3D modes, same config, same build:
[image]

SiboVG, 2024-12-20:
I think ARM64 Windows is not supported by JOGL.

jimmiedave, 2024-12-20:
Working in Windows Intel-on-ARM translation/emulation. Blinky, aliased and very slow, but working:

[image]

jimmiedave, 2024-12-20:
If JOGL is not supported on Windows ARM (and recalling today's comment that JOGL isn't maintained now), then there needs to be some kind of command decision around whether we release Windows ARM OpenRocket without 3D features. 

Ideally there'd be platform detection and removal of menu options, instead of slamming into exceptions. The software is still useful without the 3D features, and I used it for years not knowing the photo studio was there, and with non-working 3D views (15.03, I think).

jimmiedave, 2024-12-21:
> I think ARM64 Windows is not supported by JOGL.

Is there/could there be some DirectX-like layer I need to install so JOGL finds whatever it needs to talk to?

JoePfeiffer, 2024-12-21:
> If JOGL is not supported on Windows ARM (and recalling today's comment that JOGL isn't maintained now), then there needs to be some kind of command decision around whether we release Windows ARM OpenRocket without 3D features.

My first thought is we're better off having a windows ARM without 3D than no Windows ARM at all. In which case I'm remembering a little while ago I made a 32 bit Windows installer for someone on FaceBook; it also had no 3D Might as well release that too.

> Ideally there'd be platform detection and removal of menu options, instead of slamming into exceptions. The software is still useful without the 3D features, and I used it for years not knowing the photo studio was there, and with non-working 3D views (15.03, I think).

Absolutely.

closed 2026-08-24 by SiboVG