openrocket/openrocket #2639: [Bug] Possible bug - 24.12.beta.01 Cluster/Booster thrust calculations

JoeMetc, 2024-12-28:
### Describe the bug

Hi, after upgrading to the latest beta a clustered booster design I have has significantly reduced apogee, and I saw that thrust calculation had been touched in the update so was wondering if this was either a mistake or that it genuinely would fly lower and was previously inaccurate.

Cluster of Cesaroni K445 (Core) + 2x Cesaroni J420 (Boosters), Booster separation 4 seconds post launch
Previous Apogee ~3km
New Apogee ~2.15km

The reason I think this may be a bug and not just a more accurate model is that the K445 core stage alone with no boosters gets to 1.7km at 1.6kNs, wheras the full design with boosters enabled reaches only 450m higher than that even with the separation (and so lower drag) long before the end of the flight path and a hugely increased total impulse of ~3.6kNs.

Any additional details needed I'm happy to send and have email notifications set up, I can send the file privately but havent added here as it's a competition design.

### To Reproduce

1. Load design file using boosters into previous version, note down apogee
2. Load same design file into 24.12.beta.01, note down apogee
3. Apogee should be massively different if experiencing same problem
...

### Screenshots / .ork file

#### Screenshot(s):
Opening the file with the old simulation results still visible:
[image]

Updating the simulations with the new flight model:
[image]

Note that the ground hit velocity has increased (is booster mass remaining after separation?) and the hugely lower maximum acceleration (Max acceleration should be from a thrust peak of around 2250N, wheras this is showing a maximum acceleration similar to another simple rocket I have of same diameter as the core stage that only peaks around 1000N of thrust)

Motor ignition setup just for clarity:
[image]

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

24.12.beta.01

### What platform are you running on?

Windows

### Additional context

Not noticed anything like this before in any version changes over last few years. I don't have any cluster designs within a single body tube, or two stage designs, so I'd be curious if people experienced the same issues with one or both of those setups as I can't test it myself.

SiboVG, 2024-12-28:
Could you please send us the .ork file as .zip?

JoeMetc, 2024-12-28:
Hi, yep I've made an abstracted version. This file was created in the previous version of the software, to replicate open in 24.12.beta.01 then run fresh simulations to replicate the change.

[ORBoosterBugFile.zip](https://github.com/user-attachments/files/18266975/ORBoosterBugFile.zip)

JoeMetc, 2024-12-28:
I've now just scratch built a copy of this entirely in 24.12.beta.01 and the apogee is the same lower value, so not an issue to do with files changing versions to best of my checking ability

JoePfeiffer, 2024-12-31:
> Hi, yep I've made an abstracted version. This file was created in the previous version of the software, to replicate open in 24.12.beta.01 then run fresh simulations to replicate the change.
> 
> [ORBoosterBugFile.zip](https://github.com/user-attachments/files/18266975/ORBoosterBugFile.zip)

Opening this file in 23.09, it appears it was saved with 22.12.beta.01. If possible, we'd appreciate it if you could upload the 23.09 version we'd appreciate it.

JoeMetc, 2025-01-02:
Hi, yep here's the version saved in 23.09, thanks for the patience:

[ORBoosterBugFile2309.zip](https://github.com/user-attachments/files/18291710/ORBoosterBugFile2309.zip)

JoePfeiffer, 2025-01-03:
> Hi, yep here's the version saved in 23.09, thanks for the patience:
> 
> [ORBoosterBugFile2309.zip](https://github.com/user-attachments/files/18291710/ORBoosterBugFile2309.zip)

Got it, thanks.

closed 2025-01-17 by JoePfeiffer