openrocket/openrocket #2642: [Bug] The apogee height is not synchronized in the calculation and the viewport window

Errcode95, 2024-12-31:
### Describe the bug

[image]

### To Reproduce

1. Clicked the "Run calculations" button in "Flight simulation"

### OpenRocket version

24.12.beta.01

### What platform are you running on?

Windows

SiboVG, 2024-12-31:
Thank you for the report! Could you please zip the .ork file that causes this issue and post it here?

Errcode95, 2024-12-31:
[LAR-alpha.zip](https://github.com/user-attachments/files/18282553/LAR-alpha.zip)

SiboVG, 2025-01-03:
@Errcode95 thanks! Could you also share the thrust curve file for the SPAIR J111 motor please?

SiboVG, 2025-01-20:
> [@Errcode95](https://github.com/Errcode95) thanks! Could you also share the thrust curve file for the SPAIR J111 motor please?

@Errcode95?

JoePfeiffer, 2025-03-06:
It actually isn't intended to be synchronized, and in the most general case can't be. The Flight Configuration dropdown is what selects  sim results to show in the lower left hand corner.

If only one simulation is selected in the table, it would be possible to sync the flight configuration and lower left corner sim stats with that sim, but what do we do if two are selected?

(checking) It's behaved like this all the way back to 15.03 (that's the oldest runnable version I've got).

neilweinstock, 2025-05-29:
@spairtech can you please post the thrust curve file for the J111 motor?

Anyway, Based on what Joe wrote above I'm not clear if this is a bug or not.  I tend to think it is.

Generally, the flight statistics shown in the lower left of the rocket figure reflect the current selected motor configuration.  
 In the screenshot shown in the original report, there seems to be only one simulation, which suggests that there is only one motor configuration, which means that the rocket figure statistics should at least be *close* to the sim results.  The numbers shown are nowhere close.

My best guess is that this has something to do with cached sim results clashing with freshly-calculated numbers in the rocket figure display, but until we have the actual motor file and can try to reproduce the issue there's probably not much we can do.

JoePfeiffer, 2025-06-30:
I just spent a couple of minutes trying to duplicate it with a couple of random Aerotech motors, and wasn't able to. Given the user's .ork file and selecting a J500, the apogee in both places was 2022m. Switching to a J540, it changed to 2777m in both places. I checked, and I don't have "Run out-dated simulations when you open the simulation tab" checked. So I don't know how to duplicate.