openrocket/openrocket #2644: Null Pointer Exception in Appearance tab for Body Tube

The-Soupman, 2025-01-03:
-------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. Selected "Body Tube".
2. Clicked Appearance tab.
3. Unchecked Use Default in the Appearance section.
4. Tried to modify Opacity value.

If possible, please send us the .ork file that caused the bug.
- tried sending the .ORK file, but this form will not accept it for some reason

Include your email address (optional; it helps if we can contact you in case we need additional information):
soupmanrmc@gmail.com

[OR_null_ptr_exception.txt](https://github.com/user-attachments/files/18300157/OR_null_ptr_exception.txt)

SiboVG, 2025-01-03:
Hello, thank you for your report. Could you please zip the .ork file and send it here? (GitHub indeed does not accept raw .ork files)

SiboVG, 2025-01-03:
> Hello, thank you for your report. Could you please zip the .ork file and send it here? (GitHub indeed does not accept raw .ork files)

Nevermind, I'm able to replicate the bug.

The-Soupman, 2025-01-03:
[Skinwalker_v.01.ork.zip](https://github.com/user-attachments/files/18301051/Skinwalker_v.01.ork.zip)

SiboVG, 2025-01-03:
The issue is fixed now by PR #2646. Thanks again for the report and good explanation!

PR #2646 (2025-01-03): [#2644] Properly handle null events + supply source
Fixes #2644.
closed 2025-01-03 by SiboVG