openrocket/openrocket #2645: When inserting nose cone, ask where to put it

neilweinstock, 2025-01-03:
Normally I want new components added to the end of the tree... except sometimes, for whatever reason, I'm adding a nose cone after-the-fact (maybe I deleted the old one and am adding a new one, who knows).  In this case, the nose cone gets added to the back and then you must drag it to the front.  And the "match aft diameter" doesn't start working until you do that.  It's a mild annoyance.

Propose that whenever a nose cone (and *only* a nose cone) is added, if it's not the first component underneath the parent, put up a dialog asking whether to put it up front.  This will be a nice small UI convenience.

It would be possible to add an "always do this" option, or even make it a preference option, but since there are times when you actually would want to put it in back (mainly in a pod scenario) then I think it's better to just ask.  But I could be convinced otherwise.

Additional options for extra fanciness:
1) We could by default always add nose cones to the front, unless there's already a nose cone up front, in which case we add it to the back.  That would be correct the very vast majority of the time.  I can't actually think of any obvious exceptions.
2) When adding a nose cone to the back, automatically check the "flip to tail cone" box.  Since the user has already decided that the nose cone should be in back, it is *extremely* likely that they'll want it as a tail cone.  The main exception to this is if they're adding a base drag cone, which might be a sufficient reason *not* to implement this option (base drag ones are probably used more often than tail cones).

SiboVG, 2025-01-03:
Shouldn't we already have this?
[image]

neilweinstock, 2025-01-03:
I don't want to be asked with all components.  Nose cones are a special case because they are to be placed in front 99% of the time.

neilweinstock, 2025-01-03:
Since I forgot to properly account for those options, let me add this detail to the proposal: put up the nose cone dialog only if the "Always Ask" option is not selected.

I'm thinking that this should probably be another preferences option, perhaps underneath the existing first option, enabled only if "Always Ask" is not selected.  Not sure at the moment how it would be worded.

neilweinstock, 2025-01-03:
> My first thought is that maintaining consistency across component types is more important than making a special case out of one of them.

I have a different philosophy, was waiting for this comment so I could pontificate. :)

Consistency where possible and appropriate is a good thing.  However, we are not working with arbitrary generic objects.  We are working in a specific domain that we understand intimately.  Whenever we can apply that understanding to improve the experience for the user we should do so.  Because nose cones are fundamentally different from other components, in that they almost always need to go in front, only very occasionally in the back, and basically never in the middle (with possible extremely rare exceptions), we can tune the app to make them behave the way the user would expect, rather than "generically", or "exactly the same as every other component type."

Another good example within OpenRocket would be motor configuration for multistage rockets.  We know that BP boosters should almost *always* be -0.  We know that BP sustainers should almost *always* not be -0.  We can use this knowledge to help the user make the right choice more easily.  We are not doing this right now but could (and IMHO should, although exact functionality is TBD).

A less perfect example OpenRocket is the appearance panel.  It's *almost* identical for every component type, but fins and tubes have extra functionality to handle two different appearances for in/out or left/right.  That is fine and appropriate.  Again, not a perfect example but kind of.

In Windows Explorer: the "Downloads" folder has some additional chronological grouping because it reflects the typical way that that folder is used.  The Recycle Bin has extra buttons for "Empty Recycle Bin" and "Restore all items" in its toolbar, and extra custom columns in its file listing.  There is no directive that each folder in Windows Explorer must behave identically.  I mean, they *could* do that, but it would be less useful.

To summarize: I think that customizing behaviors (and possibly breaking consistency) where we have deep understanding of usage creates a better user experience.  We should of course do such things in a thoughtful way.