openrocket/openrocket #2647: Exception when opening a file created in an older version of OpenRocket

amiliv72, 2025-01-04:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. Opened an existing file.
2.
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
net.sf.openrocket.util.BugException: BUG: attempted to initialize an InertiaMatrix with a negative inertia value.
at net.sf.openrocket.masscalc.RigidBody.<init>(RigidBody.java:24)
at net.sf.openrocket.masscalc.MassCalculation.calculateStructure(MassCalculation.java:391)
at net.sf.openrocket.masscalc.MassCalculation.calculateStructure(MassCalculation.java:340)
at net.sf.openrocket.masscalc.MassCalculation.calculateStructure(MassCalculation.java:340)
at net.sf.openrocket.masscalc.MassCalculation.calculateAssembly(MassCalculation.java:291)
at net.sf.openrocket.masscalc.MassCalculator.calculate(MassCalculator.java:119)
at net.sf.openrocket.masscalc.MassCalculator.calculateLaunch(MassCalculator.java:84)
at net.sf.openrocket.gui.scalefigure.RocketPanel.updateExtras(RocketPanel.java:786)
at net.sf.openrocket.gui.scalefigure.RocketPanel.addExtras(RocketPanel.java:1084)
at net.sf.openrocket.gui.scalefigure.RocketPanel.createPanel(RocketPanel.java:461)
at net.sf.openrocket.gui.scalefigure.RocketPanel.<init>(RocketPanel.java:230)
at net.sf.openrocket.gui.main.BasicFrame.<init>(BasicFrame.java:230)
at net.sf.openrocket.gui.main.BasicFrame.open(BasicFrame.java:1364)
at net.sf.openrocket.gui.main.BasicFrame.open(BasicFrame.java:1290)
at net.sf.openrocket.gui.main.BasicFrame.openAction(BasicFrame.java:1190)
at net.sf.openrocket.gui.main.BasicFrame.openAction(BasicFrame.java:1201)
at net.sf.openrocket.gui.main.BasicFrame$27.actionPerformed(BasicFrame.java:833)
at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374)
at java.desktop/com.apple.laf.ScreenMenuItem.actionPerformed(ScreenMenuItem.java:129)
at java.desktop/java.awt.MenuItem.processActionEvent(MenuItem.java:692)
at java.desktop/java.awt.MenuItem.processEvent(MenuItem.java:651)
at java.desktop/java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:379)
at java.desktop/java.awt.MenuComponent.dispatchEvent(MenuComponent.java:367)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:776)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 23.09
OpenRocket source: default
OpenRocket location: /Applications/OpenRocket.app/Contents/Resources/app/OpenRocket.jar
User-defined thrust curves location: /Users/amiliv/Library/Application Support/OpenRocket/ThrustCurves
LAF: com.apple.laf.AquaLookAndFeel
[... 386 lines cut ...]
434650 368074.528 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
434651 368074.528 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
434652 368074.528 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
434653 368074.528 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
434654 368074.528 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
434655 368074.528 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
434656 368074.528 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
434657 368074.528 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
434658 368074.528 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
434659 368074.528 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
434660 368074.528 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
434661 368074.528 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
434662 368074.528 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
434663 368074.528 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
434664 368074.528 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
434665 368074.537 WARN (DoubleModel.java:1006) DoubleModel[constant=1.0] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@2b37298a]
434666 368074.537 WARN (DoubleModel.java:1006) DoubleModel[constant=0.0] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@37f27b0b]
434667 368074.537 WARN (DoubleModel.java:1006) DoubleModel[constant=0.2] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@3f99b03c]
434668 368074.538 WARN (DoubleModel.java:1006) DoubleModel[constant=1.0] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@d82e8e3]
434669 368074.541 WARN (DoubleModel.java:1006) DoubleModel[constant=0.0] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@5494ca94]
434670 368074.541 WARN (DoubleModel.java:1006) DoubleModel[constant=0.34519999999999995] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@5494ca94]
434671 368074.541 WARN (DoubleModel.java:1006) DoubleModel[constant=1.0] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@1559f3f8]
434672 368085.263 USER (SwingExceptionHandler.java:220) User requested sending bug report
434673 368085.275 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

amiliv72, 2025-01-04:
FWIW, I'd provide the file that caused the problem... but github doesn't let me attach ORK files.

JoePfeiffer, 2025-01-04:
You can zip the file and upload the zipped file - thanks!

amiliv72, 2025-01-04:
[saturn-v.zip](https://github.com/user-attachments/files/18306120/saturn-v.zip)

SiboVG, 2025-01-04:
This issue is fixed in 24.12.beta.01.

closed 2025-01-04 by SiboVG