openrocket/openrocket #2651: [Bug] Aerotech Motors Missing / Wrong Motors Loading

The-Soupman, 2025-01-05:
### Describe the bug

Trying to select Aerotech motors M - O impulse class, for a 75-98mm motor diameter. The M's (I think all available, I have not compared to the latest AT list) are displayed, but no N or O motors (even if selecting the + for larger cases).
[image]

### To Reproduce

1. Select Motors & Configuration tab
2. Select New Configuration
3. Clear all manufacturers and then select AeroTech
4. Adjust the range of Total Impulse for M an O (or select N by itself)
5. Select Motor Dimensions 75 to + (the 98mm selection will NOT display any motors, nor the +).

Note: If you check the box "Limit motor diameter to mount diameter", the only option becomes 75mm, even if the motor mount diameter is 98mm LOC.

...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*
[OR Bug_AT_Motor_Select_Skinwalker_v2.zip](https://github.com/user-attachments/files/18308908/OR.Bug_AT_Motor_Select_Skinwalker_v2.zip)

### OpenRocket version

24.12.beta.01

### What platform are you running on?

Windows

JoePfeiffer, 2025-01-05:
Thanks -- something seems to be happening with the length filter -- when you first enter the motor selector, what is the maximum length you see (when I first enter it, it's the infinity symbol "∞")? As long as I leave that unchanged, I see all the motors (including N and O) -- the moment I change it from that, the max case length I can see is 1000mm, and nothing I can do, including direct text entry, will change it. How close is this to what you're seeing?

The-Soupman, 2025-01-06:
Seems that is the case.

closed 2025-03-19 by JoePfeiffer