openrocket/openrocket #2653: [Bug] Force-closing Edit Simulation window retains changes without warning

eJar0k, 2025-01-05:
### Describe the bug

When closing the Edit Simulation pop-up window using the exit button or Alt-F4 after changing settings/inputs, no warning is flashed asking to confirm discarding changes, but changes are still retained. This is inconsistent with making changes and clicking cancel, which flashes the discard changes warning. What should happen when user closes the Edit Sim window, or quits OR when the Edit Sim window is open?

### To Reproduce

1. Open any OR file.
2. Create a new sim and navigate to the sims tab OR modify an existing sim.
3. Click "Edit Simulation".
4. Make a change to the simulation.
5. Close the window using the exit button or Alt-F4.
6. Re-open the window, and notice that all changes are retained.

### OpenRocket version

24.12.beta.01

### What platform are you running on?

Windows

jimmiedave, 2025-01-05:
The lack of a "Sure you want to cancel?" dialog when closing with the window gadget instead of the cancel button (in effect "Cancelling" by other means) is consistent with the behavior of the model rocket "Rocket configuration" window ("Designer" etc.).

OpenRocket may be quit with unsaved data in the "Rocket configuration" window and no "Sure you want to cancel?" dialog appears. This behavior as well is consistent with  the "Edit Simulation" window.

A couple definitions to speed this discussion:

"Dirty" - a field or window is "Dirty" if it has changed data that has not been written back to its simulation. A model is "Dirty" if it has changed data that is not written back to storage media. Closing a window that's dirty, or a file that's dirty loses the changed data.

"Clean" - the opposite of "Dirty" in both cases: There is no dirty data in a window or field (depending upon the current context) or there is no dirty data in a model: Everything has been saved to storage media and OpenRocket may be quit and re-opened, the model reloaded without loss of data.

Typically when there's the risk of data loss, user is offered a dialog which offers to preserve the edited data, or save the file. "Cancel" buttons in OpenRocket typically offer such a dialog. But in the case of window-close gadgets, OpenRocket is not affording a "Cancel" dialog when the window is dirty. Additionally, when quitting OpenRocket, an "Unsaved" dialog is afforded _if_ the model has changes from the file on disk, but _**not**_ covering the case when a window is open, but dirty (though if the _file_ is dirty at the same  time, that condition is trappped).

What should be happening is that when a window is dirty, closing the window with the close window gadget should run the "Cancel" flow.

Further, quitting OpenRocket with a window open and dirty should _also_ run the "Cancel" flow, in addition to the "File dirty" flow if the file is dirty.

jimmiedave, 2025-01-05:
Further discussion:

The Plot data and Export data tabs have similar inconsistencies, but have some differences.

The "Close" button on these tabs functions like the "Cancel" button on Simulation Settings - except that in no case does it offer a "Cancel" dialog when data is dirty.

Further, the "Plot" and "Export" buttons on their respective tabs have a **_SECRET_** function: that of a "Save" button. Whatever changes one might make to the data to plot, or the fields to export, unless one actually Plots data, no Plots data type changes will be saved. Unless one actually Exports data, no Export data type changes will be saved.

This behavior is both inconsistent, and not obvious - since the "Close" button is the only way to close the window from these tabs, sometimes (if data has been plotted or exported) the changes to the plottable/exportable data will be saved to the .ORK file when it's saved, and sometimes (if changes have been made, but _no_ data has been plotted or exported) these changes **will not** be saved to the .ORK file at save time. If changes have been made to Plottable data and Exportable data, and only **one** of Plot or Export has been used, only _some_ of the data changes will be written to the .ORK at save time.

The dialog behavior should be reviewed and made consistent.

SiboVG, 2025-01-06:
I feel like this should be configurable. I often close a window by using escape; it's become part of my workflow. I expect other people to have done the same, especially, since OpenRocket only supported the Cancel operation until very recently.

jimmiedave, 2025-01-07:
Nothing wrong with ESCAPE, nor with Close widget, nor with "Cancel". Thing is, they all represent some aspect of the flow through the window. In most other software situations, closing the window without an explicit save, whether with ESCAPE, with Close widget, or with "Cancel" is regarded as the same intent in the flow. This window (for a given tab - more tabs makes it more complicated, but consider one tab for a minute) has two intents: a primary intent of making changes to a sim and saving it, and a secondary intent of canceling those changes and leaving it as-is. More complicated flows have tertiary intents and even more (usu. too complicated by my lights...). But we just have two: Do you want to make changes, or forget those changes?

Having the window behave like other windows in other software helps the user generalize knowledge she's already got about using the computer, and thus doesn't have to "learn how to use it all over again" just to use OpenRocket. The "using the computer" part can then fade into the background and user can get busy working on a nice sim.

I hold that there should be only two ways to get out of this window (maybe 3 once we bring other tabs into the mix). Keep my changes, or Cancel my changes. There shouldn't be four (ESCAPE, Close, "Cancel", Save). In most software 3 of those (ESCAPE, Close, "Cancel") represent the same intent in closing the window, while "Save" represents the intent of saving the changes.

I recommend making this work consistently and making a decision for the user that it will work a single way (not configurable) and that way will be the way most other software works.

Not-so-humble opinion here: Configur-itis is one big reason Linux on the desktop has been very slow to be adopted indeed. That "any machine could be configured any way" means there's not a single stable target to aim for. Of course it's not the only reason. But it's a big contributor.

PR #2814 (2025-05-29): Apply uniform "cancel" behavior to all window closes and cancel button in simulations to fix #2653
in #2653 closing sim window without saving or cancelling applies sim changes quietly. This fix implements cancel-like behavior (uniformly to "cancel" button) upon closing the window without a save. The behavior honors preference setting about showing an "are you sure..." dialog, in identical manner to the "cancel" button.

`applyChangesToSimsListener()` renamed to `windowCloseCancelListener()` and targeted to new method `cancelClose()` (from prior target `copyChangesToAllSims()`, which was left a [... cut]
commit 14c3a51d (2025-05-29): Merge pull request #2814 from jimmiedave/fix2653

Apply uniform "cancel" behavior to all window closes and cancel button in simulations to fix #2653
closed 2025-05-29 by SiboVG