openrocket/openrocket #2656: [Bug] Missing truncation/rounding of long decimals in Multi Level Winds table

jimmiedave, 2025-01-05:
### Describe the bug

While "Altitude" in multi-level winds table shows as integers, clicking an altitude value shows a 12-place decimal value. It appears to be lack of rounding or truncation of the value adjusted to the user's preferred units.

### To Reproduce

1. Open "A Simple Model Rocket"
2. Change or ensure that units for "Distance" and "Velocity" are set to some imperial unit.
3. Go to "Flight simulations" tab
4. Click "Edit simulation"
5. Choose "Multi-level" in "Wind model to use"
6. Click "Add level" at least 3 times
7. Click in the table on some non-zero Altitude, Speed or Deviation
8. Note long decimal value
...

### Screenshots / .ork file

#### Screenshot(s):
[image]
[image]

### OpenRocket version

24.12.01 beta 1

### What platform are you running on?

macOS

### Additional context

macOS Sonoma 14.7.1 MacBook Pro M3 Max processor

Verified on Windows 10 too

AhanuDewhirst, 2025-02-12:
To confirm, the intended behaviour is for all values in the table to round up to the nearest significant decimal place?

[image]

SiboVG, 2025-02-12:
> To confirm, the intended behaviour is for all values in the table to round up to the nearest significant decimal place?
> 
> [image]

Ideally we would use the `round` method from `Unit`. But I'm thinking of rewriting the entire wind level table to not use a JTable but instead a custom GridLayout or something. It would make for a more elegant solution, as we can simply use existing widget code (JSpinner & UnitSelector), instead of the hacky code we have now to make the table work. Additionally, this could also help us fix the table sorting issue.

PR #2738 (2025-03-11): [#2656 & #2654] Complete rewrite of the multi-level wind UI
This PR fixes #2656 and fixes #2654 by completely rewriting the UI of the multi-level wind input. Previously, I attempted to put everything in a `JTable`. However, this solution was not flexible and required some hacks, which caused issues in units and decimal rounding. The new solution is to place the input in a dedicated dialog, which includes the wind profile visualization by default.

The multi-level panel in the sim dialog contains an overview of the wind levels:
[image]

Demo:

https://github.com/user-attachments/assets/c2132110 [... cut]
closed 2025-03-11 by SiboVG