openrocket/openrocket #2670: Eliminate "high speed deployment" warning in dual-deploy scenario

neilweinstock, 2025-01-14:
In a dual-deploy configuration (or equivalent like JLCR), a "high speed deployment" warning is inevitable when the main is deployed.  Somehow or other we need to get rid of this, since everything is operating normally.

I do not know the best solution for this, but it would be good to address it somehow.

JoePfeiffer, 2025-01-14:
It's not inevitable, and a cardboard rocket that sets it off is in some danger of a zipper -- 20 m/s is pretty quick. That said, it should be configurable.

neilweinstock, 2025-01-15:
Hmm... well, even the dual-deploy example built into OR shows the warning.  

The problem is that a configurable threshold is not really a great solution on its own, because I don't know what value to choose, and I don't want to have to change it for different rockets (again, wouldn't know what value to use).  I think there needs to be some sort of parameter attached to a recovery device and/or some heuristics that together help the app figure out how to treat it.  For instance, a plastic parachute in a BT20 rocket with apogee deploy vs. a 60 main deployed out of a fiberglass rocket while descending on a drogue.  Etc.  I don't know the answer here.

HappyCanuck, 2025-02-08:
> It's not inevitable, and a cardboard rocket that sets it off is in some danger of a zipper -- 20 m/s is pretty quick. That said, it should be configurable.

Agree with the configurable approach to avoid nuisance warnings. New users could use a default setting, but experienced users should be able to change it. Velocities between 23-30 m/s and even up to 45m/s (~150ft/s) can be acceptable for larger deployment systems.

thzero, 2026-04-19:
Gonna submit a PR for this because it was bugging me while at LDRS.

thzero, 2026-04-20:
https://github.com/openrocket/openrocket/pull/3129