openrocket/openrocket #2674: trouble with 3d

Kendrik2210, 2025-01-17:
Hi, i was usssing open rocket to make a new rocket, but when i want to see the 3d view, i can´t, can you please check if is something with my computer system or is the app?
this are the caracteristics of my computer: 

[image]

SiboVG, 2025-01-26:
@Kendrik2210 ?

dynamixbot, 2025-04-05:
@SiboVG Same issue happening with me. No error messages, just installed 23.09 version. No 3D views work, just a black box seems to be present in the place of the actual model.

SiboVG, 2025-04-05:
> [@SiboVG](https://github.com/SiboVG) Same issue happening with me. No error messages, just installed 23.09 version. No 3D views work, just a black box seems to be present in the place of the actual model.

Do you have a Windows 11 ARM machine? If so, that's not supported by the 3D engine library we use (#2622).

dynamixbot, 2025-04-05:
> Do you have a Windows 11 ARM machine? If so, that's not supported by the 3D engine library we use ([#2622](https://github.com/openrocket/openrocket/issues/2622)).

It is a Windows 11 x64 Machine with an AMD processor. OpenRocket 22.02 worked but 23.09 isn't working. Reverting back to the old version also doesn't work for me anymore.

dynamixbot, 2025-04-06:
@SiboVG any reason this could happen?

dynamixbot, 2025-04-18:
fresh install also isnt working, the issue is not with the orks either. it used to work before but when i started using openrocket again after a LOONG while it stopped working

closed 2026-08-24 by SiboVG