openrocket/openrocket #2677: [Bug] Setting launch altitude to a value higher than models can handle throws exception

JoePfeiffer, 2025-01-21:
### Describe the bug

As described in the title. The particular .ork file showing the behavior has the launch altitude set to 50,000 feet.

### To Reproduce

1. Open the .ork
2. Run the sim.
3. 
...

### Screenshots / .ork file

Two things ought to be done
1. The launch site altitude window isn't wide enough to display a four, let alone five, digit altitude. It needs to be wider
2. It should either be impossible to enter an altitude higher than the model can handle (which seems preferable to me), or there should be an error if a user tries to do it.

### OpenRocket version

24.12 beta, probably older

### What platform are you running on?

Linux

### Additional context

[2024-10-10 Level III 4.25 inch Sustainer.zip](https://github.com/user-attachments/files/18483540/2024-10-10.Level.III.4.25.inch.Sustainer.zip)

SiboVG, 2025-01-21:
> 1. The launch site altitude window isn't wide enough to display a four, let alone five, digit altitude. It needs to be wider

This was already fixed by PR #2636.

JoePfeiffer, 2025-01-22:
> > 1. The launch site altitude window isn't wide enough to display a four, let alone five, digit altitude. It needs to be wider
> 
> This was already fixed by PR [#2636](https://github.com/openrocket/openrocket/pull/2636).

Ah. Should have checked current unstable, not the beta.

AbdallahZerfaoui, 2025-01-25:
@JoePfeiffer I am currently unable to reproduce the issue in the latest unstable version. Even when setting the altitude to 1000000 meters, the simulation continues to run without any errors.

SiboVG, 2025-01-25:
> [@JoePfeiffer](https://github.com/JoePfeiffer) I am currently unable to reproduce the issue in the latest unstable version. Even when setting the altitude to 1000000 meters, the simulation continues to run without any errors.

Did you test it with the attached .ork file (`2024-10-10 Level III 4.25 inch Sustainer.zip` - you have to unzip it)? I'm still able to reproduce it with that file.

closed 2025-02-09 by SiboVG