openrocket/openrocket #2685: [Bug] Exception setting transition to 0 length

SiboVG, 2025-01-23:
### Describe the bug

Report from a user on our mailing list: 

---------- Bug report ----------
Please include a description about what actions you were performing when the exception occurred:

(You can edit text directly in this window)

1. I was changing the transition length to zero (just to see what would happen), and this error appeared

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)

---------- Exception stack trace ----------

java.lang.IllegalStateException: getRocket() called with root component Stage

at info.openrocket.core.rocketcomponent.RocketComponent.getRocket(RocketComponent.java:2276)

at info.openrocket.core.rocketcomponent.RocketComponent.addComponentChangeListener(RocketComponent.java:2552)

at info.openrocket.swing.gui.adaptors.PresetModel.<init>(PresetModel.java:49)

at info.openrocket.swing.gui.configdialog.RocketComponentConfig.<init>(RocketComponentConfig.java:154)

at info.openrocket.swing.gui.configdialog.TransitionConfig.<init>(TransitionConfig.java:52)

at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)

at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:77)

at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)

at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:500)

at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:481)

at info.openrocket.swing.gui.configdialog.ComponentConfigDialog.getDialogContents(ComponentConfigDialog.java:161)

at info.openrocket.swing.gui.configdialog.ComponentConfigDialog.setComponent(ComponentConfigDialog.java:109)

at info.openrocket.swing.gui.configdialog.ComponentConfigDialog.<init>(ComponentConfigDialog.java:62)

at info.openrocket.swing.gui.configdialog.ComponentConfigDialog.showDialog(ComponentConfigDialog.java:269)

at info.openrocket.swing.gui.configdialog.ComponentConfigDialog.showDialog(ComponentConfigDialog.java:301)

at info.openrocket.swing.gui.configdialog.ComponentConfigDialog.showDialog(ComponentConfigDialog.java:313)

at info.openrocket.swing.gui.main.RocketActions$EditAction.editComponents(RocketActions.java:586)

at info.openrocket.swing.gui.main.RocketActions$EditAction.actionPerformed(RocketActions.java:562)

at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)

at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)

at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)

at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)

at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)

at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297)

at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626)
[... 201 lines cut ...]
sun.management.compiler=HotSpot 64-Bit Tiered Compilers

user.country=US

user.dir=/Applications/OpenRocket [24.12.beta.01.app/Contents/Resources/app](http://24.12.beta.01.app/Contents/Resources/app)

user.home=/Users/christopherwhatcott

user.language=en

[user.name](http://user.name/)=christopherwhatcott

user.timezone=America/Denver

### To Reproduce

1. I was changing the transition length to zero (just to see what would happen), and this error appeared

### OpenRocket version

24.12.beta.01

### What platform are you running on?

macOS

SiboVG, 2025-01-23:
I'm not able to reproduce this bug myself...

JoePfeiffer, 2025-03-06:
I'm able to get something close (but not identical) by performing the following steps on 24.12:
1. create new rocket
2. add body tube, leave all parameters default
3. add transition, leave all parameters default
4. add body tube, leave all parameters default
5. edit transition, set length to 0

I get an exception, but instead of the illegal stage exception in the report I get 
java.lang.NullPointerException: Cannot invoke "info.openrocket.core.util.Coordinate.add(info.openrocket.core.util.Coordinate)" because "compCM" is null

I'm not able to duplicate it in current unstable (though I've got no memory of what we've changed that would have fixed it).

Shall we close with a "can't duplicate"?

SiboVG, 2025-03-09:
> I'm not able to duplicate it in current unstable (though I've got no memory of what we've changed that would have fixed it).

I think I fixed the bug you mention in #2629. I don't know if this is related. I'm okay with closing it due to lack of reproducibility.

closed 2025-03-09 by SiboVG