openrocket/openrocket #2701: [Bug] Cannot export nose cone with "export to separate files" option

ion098, 2025-02-03:
### Describe the bug

Please include a description about what actions you were performing when the exception occurred:

1. Trying to export a nose cone to obj, with the export to separate files option enabled
2. When the separate files option is not enabled, the export works fine

```
---------- Exception stack trace ----------
java.lang.RuntimeException: java.io.FileNotFoundException: C:\Users\ayaan_1_Nose Cone D.obj (Access is denied) at info.openrocket.core.file.wavefrontobj.export.OBJExporterFactory.writeObj(OBJExporterFactory.java:233) at info.openrocket.core.file.wavefrontobj.export.OBJExporterFactory.doExport(OBJExporterFactory.java:220) at info.openrocket.swing.gui.main.BasicFrame.saveWavefrontOBJFile(BasicFrame.java:1698) at info.openrocket.swing.gui.main.BasicFrame.saveAsWavefrontOBJ(BasicFrame.java:1685) at info.openrocket.swing.gui.main.BasicFrame.exportWavefrontOBJAction(BasicFrame.java:1678) at info.openrocket.swing.gui.main.RocketActions$ExportOBJAction.actionPerformed(RocketActions.java:1274) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374) at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1028) at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1072) at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389) at java.desktop/java.awt.Component.processEvent(Component.java:6391) at java.desktop/java.awt.Container.processEvent(Container.java:2266) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:775) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:747) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:744) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90) Caused by: java.io.FileNotFoundException: C:\Users\ayaan_1_Nose Cone D.obj (Access is denied) at java.base/java.io.FileOutputStream.open0(Native Method) at java.base/java.io.FileOutputStream.open(FileOutputStream.java:293) at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:235) at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:155) at info.openrocket.core.file.wavefrontobj.export.OBJExporterFactory.writeObj(OBJExporterFactory.java:230) ... 43 more

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 24.12.beta.01
OpenRocket source: default
OpenRocket location: C:\Users\ayaan.barbhuiya\AppData\Local\OpenRocket\.\jar\OpenRocket.jar
User-defined thrust curves location: C:\Users\ayaan.barbhuiya\AppData\Roaming\OpenRocket\ThrustCurves
LAF: com.formdev.flatlaf.FlatLightLaf
JOGL version: null
Current default locale: en_US
System properties:
apple.awt.application.appearance=system
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Users\ayaan.barbhuiya\AppData\Local\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Users\ayaan.barbhuiya\AppData\Local\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_users_ayaan.barbhuiya_appdata_local_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
flatlaf.uiScale=1
flatlaf.uiScale.enabled=true
install4j.appDir=C:\Users\ayaan.barbhuiya\AppData\Local\OpenRocket\
install4j.exeDir=C:\Users\ayaan.barbhuiya\AppData\Local\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.class.path=C:\Users\ayaan.barbhuiya\AppData\Local\OpenRocket\.install4j\i4jruntime.jar;C:\Users\ayaan.barbhuiya\AppData\Local\OpenRocket\.\jar\OpenRocket.jar
java.class.version=61.0
java.home=c:\users\ayaan.barbhuiya\appdata\local\openrocket\jre
java.io.tmpdir=C:\Users\AYAAN~1.BAR\AppData\Local\Temp\
java.library.path=c:\users\ayaan.barbhuiya\appdata\local\openrocket\jre\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Windows\CCM;C:\Windows\CCM;C:\Windows\CCM;C:\Users\ayaan.barbhuiya\AppData\Local\Microsoft\WindowsApps;
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=17.0.13+12-LTS
java.specification.maintenance.version=1
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=17
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=17.0.13
java.version.date=2024-10-15
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=17
java.vm.vendor=BellSoft
java.vm.version=17.0.13+12-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln10577678132775008137
line.separator=\u000d\u000a
native.encoding=Cp1252
os.arch=amd64
os.name=Windows 10
[... 238 lines cut ...]
325778 1466.792 DEBUG (BugReportDialog.java:228) Found jar file using codeSource
```

### To Reproduce

1. Open the ork file
2. Export the nose cone as an obj, with the export to seperate files option enabled

### Screenshots / .ork file

#### Screenshot(s):

[image]

#### .ork file:

[Rocket Design Template 2025.zip](https://github.com/user-attachments/files/18649139/Rocket.Design.Template.2025.zip)

### OpenRocket version

24.12.beta.01

### What platform are you running on?

Windows

SiboVG, 2025-02-04:
Thank you for the detailed report @ion098! It looks like a file permissions issue because you are trying to save the OBJ file to your C:\Users directory. Can you try saving it somewhere else (e.g. in your Downloads folder), or try running OpenRocket as an administrator?

ion098, 2025-02-04:
@SiboVG 

I'm using a school computer to run OpenRocket right now, so I can't test with administrator mode unfortunately. I did try saving it in Downloads and Documents, but it always complains about the same path: `C:\Users\ayaan_1_Nose Cone D.obj` (without the "export to separate files" option, everything works fine)

ion098, 2025-02-05:
@SiboVG 

I believe I've found the issue: 

When the export as separate files option is checked, we hit this code path:

https://github.com/openrocket/openrocket/blob/060e6d0411ae4e5b0ea41f17259a70ac993a39e1/core/src/main/java/info/openrocket/core/file/wavefrontobj/export/OBJExporterFactory.java#L163-L165

`this.file.getAbsolutePath` returns `"C:\Users\ayaan.barbhuiya\Downloads\file_name.obj"`. The issue lies in the implementation of `FileUtils.removeExtension`:

https://github.com/openrocket/openrocket/blob/060e6d0411ae4e5b0ea41f17259a70ac993a39e1/core/src/main/java/info/openrocket/core/util/FileUtils.java#L46-L52

This function implicitly assumes that file paths only contain one `.`. As a result, `removeExtension` ends up returning `"C:\Users\ayaan"`, instead of `"C:\Users\ayaan.barbhuiya\Downloads\file_name"` like it should. The end result is that we end up with a file path of `"C:\Users\ayaan_1_Nose Cone D.obj"`, instead of the proper path, `"C:\Users\ayaan.barbhuiya\Downloads\file_name_1_Nose Cone D.obj"`, leading to the permission error.

SiboVG, 2025-02-05:
> [@SiboVG](https://github.com/SiboVG)
> 
> I believe I've found the issue:
> 
> When the export as separate files option is checked, we hit this code path:
> 
> [openrocket/core/src/main/java/info/openrocket/core/file/wavefrontobj/export/OBJExporterFactory.java](https://github.com/openrocket/openrocket/blob/060e6d0411ae4e5b0ea41f17259a70ac993a39e1/core/src/main/java/info/openrocket/core/file/wavefrontobj/export/OBJExporterFactory.java#L163-L165)
> 
> Lines 163 to 165 in [060e6d0](/openrocket/openrocket/commit/060e6d0411ae4e5b0ea41f17259a70ac993a39e1)
>  if (exportAsSeparateFiles) { 
>      String path = FileUtils.removeExtension(this.file.getAbsolutePath()) + "_" + groupName + ".obj"; 
>      objFileMap.put(path, obj); 
> 
> `this.file.getAbsolutePath` returns `"C:\Users\ayaan.barbhuiya\Downloads\file_name.obj"`. The issue lies in the implementation of `FileUtils.removeExtension`:
> 
> [openrocket/core/src/main/java/info/openrocket/core/util/FileUtils.java](https://github.com/openrocket/openrocket/blob/060e6d0411ae4e5b0ea41f17259a70ac993a39e1/core/src/main/java/info/openrocket/core/util/FileUtils.java#L46-L52)
> 
> Lines 46 to 52 in [060e6d0](/openrocket/openrocket/commit/060e6d0411ae4e5b0ea41f17259a70ac993a39e1)
>  public static String removeExtension(String fileName) { 
>  	String[] splitResults = fileName.split("\\."); 
>  	if (splitResults.length > 0) { 
>  		return splitResults[0]; 
>  	} 
>  	return fileName; 
>  } 
> 
> This function implicitly assumes that file paths only contain one `.`. As a result, `removeExtension` ends up returning `"C:\Users\ayaan"`, instead of `"C:\Users\ayaan.barbhuiya\Downloads\file_name"` like it should. The end result is that we end up with a file path of `"C:\Users\ayaan_1_Nose Cone D.obj"`, instead of the proper path, `"C:\Users\ayaan.barbhuiya\Downloads\file_name_1_Nose Cone D.obj"`, leading to the permission error.

Thank you for the analysis, I was instead also looking at that code last night. We'll have it fixed soon!

PR #2705 (2025-02-08): [#2701] Make remove file extension method robust against periods in the file path
This PR fixes #2701. The issue was with the `removeExtension` method in `FileUtils` which was not robust against file paths with a period in it.
closed 2025-02-08 by SiboVG