openrocket/openrocket #2706: Drop down box repeats next to last entry

sjh114961, 2025-02-05:
### Describe the bug

Drop down box repeats next to last entry in nose cone custom selection

[image]

### To Reproduce

1. Select 8 nose cones for custom nose cones
2. 
3. 
...

### Screenshots / .ork file

[image]

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

OpenRocket 24.12.beta.01

### What platform are you running on?

Windows

SiboVG, 2025-02-09:
Looks like it is part-dependent, e.g. PNC-60L and PNC-60MS are duplicated, regardless of how many custom parts are selected (only adding PNC-60L already triggers the bug).

[image]

SiboVG, 2025-03-22:
Okay, so I found the issue. The duplicate parts are parts that are present in both the new [openrocket-database](https://github.com/openrocket/openrocket-database) and the internal legacy parts database (`core/src/main/resources/datafiles/components/internal/Estes-legacy.orc`). I actually don't know the purpose of the legacy database. @JoePfeiffer, do you know? Are not all the parts of the legacy database part of the openrocket-database? If not, I think the best solution to this issue is to remove entries from the legacy database that are already part of the openrocket-database.

JoePfeiffer, 2025-03-22:
> Okay, so I found the issue. The duplicate parts are parts that are present in both the new [openrocket-database](https://github.com/openrocket/openrocket-database) and the internal legacy parts [...]

Last I looked, there were parts in the legacy database that are not in the new database. Yes, the best solution is to remove parts from the legacy database when they also appear in the new database; would somehow indicating a part is from the legacy database when it appears in the dropdown? Maybe just text like _(legacy)_?

PR #2758 (2025-03-23): [#2706] Remove duplicate legacy database parts
This PR fixes #2706 by removing parts from the legacy database that are already part of the [openrocket-database](https://github.com/openrocket/openrocket-database) library.
closed 2025-03-23 by SiboVG