openrocket/openrocket #2707: Text too far to left in dialog boxes for Rocket configuration entry

sjh114961, 2025-02-05:
### Describe the bug

Text too far to left in dialog boxes for Rocket configuration entry See screen shot

### To Reproduce

1. Simply type text in boxes
2. 
3. 
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

[image]

### OpenRocket version

OpenRocket 24.12.beta.01

### What platform are you running on?

Windows

SiboVG, 2025-02-05:
I'm sorry, I don't understand the issue, could you please clarify?

sjh114961, 2025-02-05:
Look where I placed the RED "T". It is one character from the left side.

SiboVG, 2025-02-05:
> Look where I placed the RED "T". It is one character from the left side.

I saw the 'T', but I still don't know what it means. Is this something you added to your screenshot or is it an artifact generated by OpenRocket?

Is your point that there's too much white space to the left of your text entries (`Stephen J. Heilman` and `Original: 02/05/2025`)?

sjh114961, 2025-02-06:
I added it to show that should be moved over by 1 character.