openrocket/openrocket #2714: Cannot add user created extensions in Simulation options

HappyCanuck, 2025-02-08:
### Describe the bug

OpenRocket doesn't detect extension files that are placed in the Plugins directory.
[image]

### To Reproduce

1. Place a jar file extension in the OpenRocket Plugins directory: C:\Users\ImAUserAccount\AppData\Roaming\OpenRocket\Plugins

2. Reload OpenRocket
3. Open or Edit a Simulation window to Add extension
4. File a bug report
...

### Screenshots / .ork file

### OpenRocket version

24.12 beta 01

### What platform are you running on?

Windows

### Additional context

Under Preferences, a new field was added in 24.12 beta01 to specify directories for User-defined component presets. Maybe a similar field for Plugins is missing?

[image]

SiboVG, 2025-02-08:
This has to do with #2676. We renamed the package name of the OpenRocket codebase (it was still using an ambiguous, old name), which breaks compatibility with old plugins that still use the `net.sf.openrocket` reference instead of the new `info.openrocket.core` and `info.openrocket.swing`.

closed 2025-02-08 by SiboVG

HappyCanuck, 2025-02-09:
Thanks for the quick clarification Sibo -
May I suggest adding a "Breaking Changes" section in the Release notes for 24.12? I can open a new issue requesting this if you would prefer.