openrocket/openrocket #2721: [Feature Request] Use thrustcurve.org API to determine motor manufacturers

JoePfeiffer, 2025-02-19:
**Is your feature request related to a problem? Please describe.**
When new motor manufacturers are added to thrustcurve.org, we don't see them unless we manually add them to the manufacturers list in SerializeThrustCurveMotors.java

**Describe the solution you'd like**
thrustcurve's API is able to provide a list of manufacturers (along with impulse classes, diameters, etc). We should use that to populate our options here.

senft-research, 2025-05-08:
I see you changed the names of the manufacturers. After #2799 is resolved, would you like me to look further into this and make this more dynamic? 

Or would you rather my efforts go into bug squashing like I've attempted with #2777 ? Happy to do whatever is more in line with current priorities.

JoePfeiffer, 2025-05-08:
My suggestion is you look at this and I'll look at the warnings bug.

senft-research, 2025-05-08:
Sounds like a plan to me :)

senft-research, 2025-05-08:
As clarification, would it be prudent to wait until the current PR is merged? I can imagine the risk of Merge Conflicts being higher if I start messing with the way manufacturer names work in another branch.

PR #2806 (2025-05-24): Motor Manufacturer determination via ThrustCurveAPI
Retrieves the Manufacturers directly from an GET call to the ThrustCurveAPI. 

Removes need to hardcode a manufacturers String array within the Serialization Code. 

_Dev Note: Not sure why my other commits were not verified via my GPG key, issue seems to be resolved._

Closes #2721
closed 2025-05-24 by JoePfeiffer