openrocket/openrocket #2724: bug report

kta0120, 2025-02-22:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.Added streamer
2.It was added above the nose cone, so it was moved below the nose cone.
3.I was asked to report a bug
4.I reported it because I had free time.

Is this okay?

[bugreport.txt](https://github.com/user-attachments/files/18918295/bugreport.txt)

neilweinstock, 2025-02-24:
Thank you for the report.  Can you attach the ORK file that caused the bug?

kta0120, 2025-02-24:
I don't know if it's compressed correctly, but this is it.
[rocket1.zip](https://github.com/user-attachments/files/18936059/rocket1.zip)

JoePfeiffer, 2025-02-24:
The .ork file looks OK, and contains the streamer. I'm having a hard time duplicating the error.

kta0120, 2025-02-24:
The procedure where the error occurred was to select the nose cone and add a streamer to it. The streamer's position was strange, so I moved it to the body tube directory? and an error occurred.
This may be because the language setting is set to Japanese.

SiboVG, 2025-02-24:
> This may be because the language setting is set to Japanese.

I set my UI to Japanese (interesting experience!), but can't trigger the bug. @kta0120 are you able to reliably reproduce this bug, or did it only happen once?

JoePfeiffer, 2025-02-25:
This appears to be a duplicate of #2418 which has been fixed for the next version.

closed 2025-02-25 by JoePfeiffer

kta0120, 2025-02-25:
The bug can be reproduced any number of times.
I certainly hadn't updated the version. Sorry for taking up your time.