openrocket/openrocket #2731: Possible inaccuracy in calculation of atmosphere properties

Vaokon, 2025-03-03:
From what I can see, the atmospheric conditions are calculated according to the ISA model using geopotential altitudes. (ExtendedISAModel.java)

However, it seems the simulation operates using the geometric altitude. (AbstractSimulationStepper.java)

This geometric altitude is then passed into the atmospheric model, which expects geopotential altitude.

If I am right, this could be a source of error, especially at higher altitudes.
(the difference is about 20 meters at 11 km altitude and 162 m at 32 km)

This could be solved by converting the geometric altitude into the geopotential altitude using the formula shown after the main table in this article:
https://en.wikipedia.org/wiki/International_Standard_Atmosphere
(It also seems that the last layer's altitude is set to the geometric altitude instead of the geopotential one, ie 84852 instead of 86000)

Vaokon, 2025-03-03:
Seems like it could be changed in either of those places?

[image]

This one seems like a better idea. 
I think it would have to be converted in many places in this file. Like every time a function receives the altitude as a parameter.
[image]

JoePfeiffer, 2025-03-03:
162 meters at 32 km is about 0.5%. I'd say that isn't worth worrying about, except somebody implemented coriolis forces at some point... 

It seems like the best place to make the change would be within the atmospheric conditions calculators, so externally it still uses geometric altitude.

Vaokon, 2025-03-04:
Yeah. not a big deal.

In the calculator makes the most sense imo.

BTW ignore this, wikipedia had them the wrong way around. 
> (It also seems that the last layer's altitude is set to the geometric altitude instead of the geopotential one, ie 84852 instead of 86000)