openrocket/openrocket #2737: [enhancement] Get winds and temperature aloft forecasts from open-meteo.com

JoePfeiffer, 2025-03-08:
https://open-meteo.com offers winds- and temperature- aloft forecasts, with a free web API. We should make it possible for users to obtain this data as needed.

Note that this implies an enhancement to the existing wind levels for temperature (replacing the ISA temperature model) as well.

The forecast is updated hourly. They request their API only be used for a maximum of 10,000 API calls per day. While I doubt our total user base would come anywhere near this limit, we ought to only allow a forecast every hour to limit the load on their servers.

kjhambrick, 2025-03-15:
Open-Meteo also provides the pressure gradient so air density and mach number aloft could be calculated from the provided data sets.

JoePfeiffer, 2025-03-15:
> In addition to wind speed and direction and temperature, Open-Meteo also provides the pressure gradient so air density and mach number aloft could be calculated from the provided data sets.

The more I think about what we can get from open-meteo the more excited I get.