openrocket/openrocket #2741: [Bug] Exception when opening this file

SiboVG, 2025-03-09:
### Describe the bug

User RussellCasse reported a bug when opening the attached file:
Unable to open file 'NEVIS-WM3-2-stage.ork': Exception loading stream: BUG: Message with id 9b1d8c35-c433-46e8-b1d2-adc45712df52 not found

### To Reproduce

1. Open the file

### Screenshots / .ork file

#### .ork file:
[NEVINS-WM3-2-stage.ork.zip](https://github.com/user-attachments/files/19147675/NEVINS-WM3-2-stage.ork.zip)

### OpenRocket version

unstable

### What platform are you running on?

Windows

### Additional context

Not present in 23.09

SiboVG, 2025-03-09:
The problem is caused by the `simwarn` event which has an id that does not correspond to an existing warning.

```xml
<databranch name="Booster" optimumAltitude="966.0301674012272" timeToOptimumAltitude="14.592647385008886" types="Time,Altitude,Position East of launch,Position North of launch,Vertical velocity,Lateral distance,Total velocity,Lateral direction,Vertical acceleration,Lateral velocity,Total acceleration,Lateral acceleration,Latitude,Longitude,Angle of attack,Roll rate,Pitch rate,Yaw rate,Vertical orientation (zenith),Lateral orientation (azimuth),Mass,Motor mass,Longitudinal moment of inertia,Rotational moment of inertia,Gravitational acceleration,CP location,CG location,Stability margin calibers,Thrust,Thrust-to-weight ratio,Drag force,Drag coefficient,Friction drag coefficient,Pressure drag coefficient,Base drag coefficient,Axial drag coefficient,Normal force coefficient,Pitch moment coefficient,Yaw moment coefficient,Side force coefficient,Roll moment coefficient,Roll forcing coefficient,Roll damping coefficient,Pitch damping coefficient,Wind velocity,Air temperature,Air pressure,Air density,Speed of sound,Mach number,Reynolds number,Reference length,Reference area,Simulation time step,Computation time,Coriolis acceleration">
          <event time="0" type="ignition" source="5e32c22e-a5f7-47ae-9142-e42b53d534ab"/>
          <event time="1.274" type="burnout" source="5e32c22e-a5f7-47ae-9142-e42b53d534ab"/>
          <event time="9.274" type="ejectioncharge" source="edc577df-28d1-4670-bbb8-536707ac0cb4"/>
          <event time="9.274" type="stageseparation" source="edc577df-28d1-4670-bbb8-536707ac0cb4"/>
          <event time="9.277" type="simwarn" id="9b1d8c35-c433-46e8-b1d2-adc45712df52"/>
```

JoePfeiffer, 2025-03-09:
I think this is a duplicate of #2694. I'll make sure.

SiboVG, 2025-03-09:
> I think this is a duplicate of [#2694](https://github.com/openrocket/openrocket/issues/2694). I'll make sure.

Ah yes, sorry, my local unstable branch was outdated, this is already fixed in the latest unstable.

nevinsb, 2025-03-10:
Thanks everyone for looking at it!