openrocket/openrocket #2750: [Bug] <insert your title>OR 2412 RC1 - Sim > Plot > Stability -vs- Time - Noise during thrust phase

kjhambrick, 2025-03-15:
### Describe the bug

Built 2412 - RC1 on 2025-03-11 at 05:54 CST / 11:54 GMT via `git clone` and gradle on Slackware Linux 15.0 ( 64 bit ) with openjdk17-17.0.14_7

When I simulate Stability -vs- Time, there is a lot of noise in the plot during the thrust phase.

Exported Plot below.

Is there a gradle build log that might help ?

Thanks everybody !

-- kjh

### To Reproduce

1. Build 2412 RC1and start it up
2. Load an existing Model
3. Simulate Stability -vs- Time
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the s

[image]

[image]

[MAUDE-38-1200-loki-md-hed-power-perch-4-x-2.000-span-m2.50-fins-2412-RC01.ork.zip](https://github.com/user-attachments/files/19261130/MAUDE-38-1200-loki-md-hed-power-perch-4-x-2.000-span-m2.50-fins-2412-RC01.ork.zip)

creenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

24.12 RC1

### What platform are you running on?

Windows

JoePfeiffer, 2025-03-15:
Is the .ork file you posted the same rocket as the screen shots? There's a lot of internal structure in the .ork file (overlapping the J510 motor) that I don't see in the screen shots.

Regardless, I'll wait for other people's opinions, but I don't think this is a bug. You've got a rocket with a really high longitudinal moment of inertia and relatively small fins; this means the angle of attack is doing quite a bit of oscillation before settling down. Non-zero angle of attack moves the center of pressure forward; see this plot:
[image]
(it actually displays the magnitude of the angle of attack, giving the illusion that it's bouncing off 0 instead of oscillating).
As the oscillations die down the stability also settles down.

(you will be happy to know that the most recent unstable has fixed the "zillions of supersonic warnings" bug)

kjhambrick, 2025-03-16:
Thanks Joe !

Yes, this rocket is pretty long to accomodate a Loki 38-1200 motor so off-axis disturbances ( like the 12mph wind ) will be slow to decay.

But !

I ammended my ticket entry after I saw that the 'noise' was actually a series of Mach 1.1 warning markers :)

I acually expected the wiggles -n- squiggles in the stability -vs- time plot but I didn't understand that the greenish 'noise' was actually a series of warnings.

I'll `git` a new set of source code and build and try it.

Thanks again !!!

-- kjh

kjhambrick, 2025-03-16:
Joe --

I set aside the old 24.12 RC1 code and reran `gradle clean check build dist`

The internal structure that overlaps the motor that you see in the .ork is due to three variations of an Oatey min diameter motor retainer that I included in the model

[image]

Note that the x-long and long have mass and cg overrides -- they're set to 0.00 and the sub-components override is clicked.

When I open the.ork, I use the new 24.12 feature to hide the Oatey assemblies that don't fit the motor.

Note that only the 'std' Oatey is visible in the screen shot.

I believe that explains the apparent difference between the .ork and the screen shot ?

I still see the Mach 1.1 warnings though:

[image]

Did I download the wrong code via `git clone https://github.com/openrocket/openrocket.git` ?

Or maybe I misunderstood what you were saying about removing the zillions of supersonic warnings ?

Thanks again !

-- kjh

JoePfeiffer, 2025-03-17:
Try doing a 
git pull
to make sure you're up to date with the repository. Fixing the "multiple warnings" was just last Wednesday.
If you then do a
git log
one of the commits should be

commit b7c0edc0fc0919f5213dd842c91a8700080c2ddd
Merge: 2e8b0c9d1 9f3fa504c
Author: Joe Pfeiffer <joseph@pfeifferfamily.net>
Date:   Wed Mar 12 14:40:37 2025 -0600

    Merge pull request #2746 from JoePfeiffer/open-warnings-again
    
    If a warning isn't actually added to the warning set, don't put a new flight event for it.

Assuming that's in there, recompiling should give you a version that doesn't produce all the extra warnings.

kjhambrick, 2025-03-17:
Joe --

I did `git pull` and I got:  Already up to date.

So I did `git log |less` and I do see your commit:

``commit b7c0edc0fc0919f5213dd842c91a8700080c2ddd
Merge: 2e8b0c9d1 9f3fa504c
Author: Joe Pfeiffer <joseph@pfeifferfamily.net>
Date:   Wed Mar 12 14:40:37 2025 -0600

    Merge pull request #2746 from JoePfeiffer/open-warnings-again
    
    If a warning isn't actually added to the warning set, don't put a new flight event for it.
``

However I still see all the warnings.

Thoughts ?

-- kjh

JoePfeiffer, 2025-03-17:
> Thoughts ?
> 
> -- kjh

I'm mystified -- I've got no idea how it can be behaving differently for you than for me.  I've put my .jar file on google Drive; try grabbing it and see how it works for you.

https://drive.google.com/file/d/16qsDAH03PRWonvS5812JLMVuCJyhU-6R/view?usp=sharing

kjhambrick, 2025-03-17:
Joe --

I see the same thing in your OpenRocket-24.12.RC.01.jar

[image]

But then I saw the [ ] Warning Check Box on the Plot Setup:

[image]

I can 'clean up' the plots by disabling Show Warnings:

[image]

That'll work for me !

Thanks !

-- kjh

JoePfeiffer, 2025-03-17:
I'm glad you've found a workaround that helps you, but there still seems to be a really screwy bug I need to track down....

You are re-running the simulations, aren't you?

kjhambrick, 2025-03-18:
Joe --

Yes, I rerun the sims frequently.

Are you able to dupe the issue ?

Thanks for looking into this !

-- kjh

JoePfeiffer, 2025-03-18:
> Are you able to dupe the issue ?

No, that's what has me so confused. When I ran an older version I saw the issue; when I run the .jar I sent you I don't. Could you put your build some place I can get it, so I can see if that duplicates it for me?

kjhambrick, 2025-03-18:
I'll try to put it on google drive and let you know.

JoePfeiffer, 2025-03-18:
Same thing running yours as running mine -- I open the .ork, plot the first sim, get a million supersonic warnings. Run the sim, plot again, I just get one warning when it first goes supersonic. I'm baffled.

kjhambrick, 2025-03-19:
That's weird, Joe.

I can live with unchecking the [ ] Warnings Check Box

Thanks for all the time you spent helping me with this.

-- kjh

SiboVG, 2025-03-22:
I can replicate this issue on the latest unstable:

[image]

However, rerunning the sim restores normal behavior:

[image]

When I then save the .ork and re-open it, I again have normal behavior (only a few warnings).

So, is this still a bug? I think the .ork was saved in a version of OR that had the original bug, but it appears that it's fixed in the latest unstable.

JoePfeiffer, 2025-03-22:
> So, is this still a bug? I think the .ork was saved in a version of OR that had the original bug, but it appears that it's fixed in the latest unstable.

I'm seeing the same behavior you are, and I don't understand why the user who reported the original bug is still seeing it.

closed 2025-03-22 by JoePfeiffer