openrocket/openrocket #2755: [Enhancement] Make it more clear that simulation can not be run because it has no motors

SiboVG, 2025-03-22:
When you create a sim with no motors, there is no feedback that the simulation can't run _because_ it has no motors. The only feedback the user receives is "Errors in simulation prevent running". Not very useful...

Example:

[image]

There's not even a warning. In 23.09, if you ran a simulation with no motors, at least you got an error message:

[image]

This is not the case anymore in 24.12... This tempts me to file this as a release blocker.

JoePfeiffer, 2025-03-22:
I'm not seeing this. Running both 24.12.beta.01 and current unstable, if I open the "Simple Model Rocket" example, create a new configuration but don't select any motors in it, and run the corresponding sim, I get:

[image]

SiboVG, 2025-03-22:
Yes, in the example above, I used a blank design (no components). For normal designs, the error is indeed correct.