openrocket/openrocket #2756: [Feature Request] Motor Size in Flight Sims Tab

Flippit1303, 2025-03-22:
Having motor specs with the flight sims allows you to sort by either motor diameter, case size, RMS or DMS.  You can create a "master list" of all motors that work for that rocket and have a reference sheet to what size hardware you need especially if you use motor adapters.

Example: You have a 54-1706 that fails in some way that you can't use it anymore. Without digging through all the motor specs, you can quickly see what motors you Can/can't use with that case anymore.

JoePfeiffer, 2025-03-22:
That's a good idea -- it ties up well with something else that's a medium/long term goal, which is to add a *bunch* more columns to the flight sim table, and let the user select which ones they want to have visible.

SiboVG, 2025-03-23:
Superseded by #2704.

closed 2025-03-23 by SiboVG