openrocket/openrocket #2762: [Feature Request] Store motor data in .ork files

JoePfeiffer, 2025-03-24:
At present, .ork files only contain motor information sufficient to look up a motor in the database and don't contain the actual thrustcurve.  This causes two problems: 

First, the motor "digest" (a hash function) used to match the motor against candidates in the database is brittle, and it's not uncommon to locate a motor in the database and have the digests not match, in which case the user is presented with an error dialog saying the motor's thrustcurve has changed.

Second, when a user uses a motor that is not in the database (for instance, an experimental motor or a motor that has been certified since our last release) and wishes to share their .ork file with another user, they need to supply the thrustcurve file separately. 

If we were to save all of the data for a motor in the .ork file these problems would be resolved with a really minimal increase in the size of the .ork file.

SiboVG, 2025-03-24:
Duplicate of #1838.

closed 2025-03-24 by SiboVG

JoePfeiffer, 2025-03-24:
I thought I'd suggested that previously.... no idea why a search didn't turn it up. Thanks.