openrocket/openrocket #2769: [Bug] OR 24.12 RC-01 - Export Sim Data Latitude and Longitude always print Radians

kjhambrick, 2025-03-29:
### Describe the bug

Exporting Sim Data from OR 2412 RC01 and Latitude and Longitude always print signed Radians

The numbers look correct, However, the units are incorrect.

Example:  

Set up Site Lat and Long as:

Latitude ..........: 30.61473
Longitude .......: -97.49701

Time (s) = 0.000000 row shows:

Latitude (Â° N) ..: 0.534328
Longitude (Â° E) : -1.701644

Converting the numbers in the .csv from radians to degrees gives me:

Latitude ..........:  30.614739
Longitude .......: -97.497020

About what I started with in the Launch Conditions dialog.

Maybe while fixing the conversion, allow for 'signed decimal degrees' for latitude and longitude instead of Deg { N,S,E,W } ?

I am still going thru the exported data so this report may be incomplete ...

Thanks, all'Y'all are awesome !

-- kjh

Attached:  Exported .csv file

[tp-h180w-test-export.csv](https://github.com/user-attachments/files/19521124/tp-h180w-test-export.csv)

### To Reproduce

1. Run a Sim,
2. Set Latitude and Longitude units to 'Degrees',
3. Export Data to a .csv file,
4. Lat and Long are always radians in the .csv file.

### Screenshots / .ork file

#### Screenshot(s):
[image]

#### .ork file:
[vul-48-2412-RC01.ork.zip](https://github.com/user-attachments/files/19521121/vul-48-2412-RC01.ork.zip)

### OpenRocket version

24.12 RC-01

### What platform are you running on?

Linux

senft-research, 2025-05-12:
Can you clarify what you mean by what you mean by you "setting the longitude and latitude to degrees" as, on review, longitude and latitude are purely set to terms of N, E, S, W, where as it is the lateral and longitudinal orientations that can be set to degrees / radians. 

Longitude and Latitude themselves are always recorded as radians from investigation into the matter. 

If I am mistaken in this please elaborate so I can look into this further :)

SiboVG, 2025-05-12:
Fixed by #2793.

closed 2025-05-12 by SiboVG

kjhambrick, 2025-05-12:
> Can you clarify what you mean by what you mean by you "setting the longitude and latitude to degrees" as, on review, longitude and latitude are purely set to terms of N, E, S, W, where as it is the [...]

Was your question meant for me ?

I have never seen latitude and longitude reported to human readers as radians, nor have I ever seen a system that expects a human being to enter ( lat,long)  as radians

Of course, all the trig under the hood does need to be performed with radians.

Latitude is simplest IMO when the readings are normalized to decimal degrees in the domain { -90 .le. latitude .le. 90 }

The standard for GPS positioning is normalized decimal degrees.

Was for what's what, -90 degrees latitude is the south pole and +90 degrees is the north pole and 0 degrees is the equator.

And longitude is normalized on the domain { -180 .le. longitude .le. 180 }

0 degrees is the prime meridian in Greenwich, England.

Positive longitudes are east of Greenwich and negative longitudes are west.

Does this help ?