openrocket/openrocket #2770: open rocket running into problems

John-design-345, 2025-03-29:
Always when i select or edit a component on the rocket then it shows a bug.

JoePfeiffer, 2025-03-29:
> Always when i select or edit a component on the rocket then it shows a bug.

Please, we need a **_bunch_** more information before we can help you.
1. What bug is it reporting? Is it showing a stack trace? Please add that to your report.
2. What version of OpenRocket?
3. Do you have an existing .orc file? If so, please zip it up and add that as well.

jimmiedave, 2025-04-08:
> 3. Do you have an existing .orc file? If so, please zip it up and add that as well.

.ork file, I think we're talking.

JoePfeiffer, 2025-04-08:
Yes, thank you.

JoePfeiffer, 2025-04-10:
No response from original reporting user.

closed 2025-04-10 by JoePfeiffer